dreemurrs-embedded/Pine64-Arch #105: GPS not works

juanantoniomosquera, 2020-11-30:
Hello,

I try GPS and... not works.

I try:
[image]

And:
[image]

But... with data mobile and gps "enabled" not geolocalization.

Thanks.

Lin-Admin, 2021-01-16:
With latest system updates 5.10.6-2-danctnix, gps does not deliver any location information:
`mmcli -m any --location-get
  3GPP |      operator code: 228
       |      operator name: 3
       | location area code: FFFE
       | tracking area code: BB84
       |            cell id: 002BD002
`

juanantoniomosquera, 2021-01-16:
with last update... location-get works... but.. without wifi enable... not works gps.

Lin-Admin, 2021-01-19:
@juanantoniomosquera:
With my phone it makes no difference whether wifi is on or off: GPS looks dead...

abysso2, 2021-02-01:
> With latest system updates 5.10.6-2-danctnix, gps does not deliver any location information:
> `mmcli -m any --location-get
> 
> 3GPP | operator code: 228
> | operator name: 3
> | location area code: FFFE
> | tracking area code: BB84
> | cell id: 002BD002
> `

Hi there,
what is the expected ouptput to prove the location services?

A.

juanantoniomosquera, 2021-02-03:
[image]

Lin-Admin, 2021-02-03:
`GPS | nmea: $GPGSA,A,1,,,,,,,,,,,,,,,,32

      | $GPRMC,,V,,,,,,,,,,N53

      | $GPVTG,,T,,M,,N,,K,N*2C

      | $GPGGA,,,,,,0,,,,,,,,*66`

Above output proves that the GPS does not detect any satellites. For a location fix you need at least 4, but below 6 the accuracy will not be good.

juanantoniomosquera, 2021-02-03:
NOW

[image]

But... gnome maps not works.

jackghg, 2021-02-10:
I think that the phone antenna is real bad!

abysso2, 2021-03-01:
@Danct12 : Any idea how to solve this. Smartphone without GPS is not thaaaaat good as a daily driver ...

jackghg, 2021-03-24:
A possible reason could be that the antenna is bent around the corner!
But idk how to move the antenna (that is a sticker) to test... without breaking everything.

abysso2, 2021-03-24:
Switched to latest mobian and tested - gps works there ...

jackghg, 2021-03-25:
really ? how long to get  the position (without internet) ?

abysso2, 2021-03-25:
Less than 30 Seconds to get the signal, but the (in the beginnig) BAD accuracy increases within 5 minutes ...

jsarkari01, 2022-03-24:
I can get location to work in GNOME maps app however that is a full desktop app and it runs off the page in portrait so better in landscape. However the location button in notifications is always off ie you cannot toggle on or off their.

hamblingreen1, 2022-09-06:
location is functional via Mepo maps on sxmo.

jacobhawkins495, 2024-08-17:
I cannot for the life of me get my PinePhone 1.2 (Manjaro Edition) to obtain a GPS fix. It is running  Pine64-Arch version `
archlinux-pinephone-phosh-20240326.img.xz` and ADSP EG25GGBR07A08M2G_30.006.30.006 (via `AT+QMGR`). The system has been updated via `pacman -Syu` and is running the latest available software as of this posting. The device can see loads of satellites as shown by `mmcli -m any --location-get` once `mmcli -m any --location-enable-gps-nmea --location-enable-gps-raw --location-enable-agps-msa` is run, however as indicated by the `GPGGA` line and lack of output from GPS raw, the phone never obtains an actual fix and won't ever geolocate. Included is a sample output from `mmcli -m any --location-get` showing the satellite lists and lack of fix. It does not matter how long I wait or whether I'm in a building or outside on a cloudless day; the phone will never obtain a fix and a geolocation.

[nmea-messages.log](https://github.com/user-attachments/files/16643390/nmea-messages.log)