dreemurrs-embedded/Pine64-Arch #106: Block sleep/suspend when playing music

mogwa1, 2020-12-04:
Currently, the phone will always suspend after being inactive for x minutes (default=5) when running on battery.
This is quite inconvenient when e.g. listening to music in the background.

I think on desktop linux it's possible to skip suspending as long as media is playing, so would it also be possible here?

JasonG-FR, 2020-12-11:
I was using caffeine-ng from the AUR to do this until now. It doesn't work anymore since Python was updated to 3.9.

It was far from being perfect, you had to manually launch the preference window by using the `caffeine -p` and also make sure that it's launched at boot (I used gnome-tweaks to do this).

mogwa1, 2021-05-22:
After having developed a kirigami-based podcast player myself (invent.kde.org/plasma-mobile/kasts), it is now clear that this is actually something that needs to be implemented by each and every application.  (DBus call to inhibit power management to kick in while playing)

It just happens to be the case that none of the gnome-based applications (gnome podcasts, lollypop) have this implemented.
So I guess there is nothing you can do about this.  I'll close this.

closed 2021-05-22 by mogwa1