dreemurrs-embedded/Pine64-Arch #111: Anbox broken after 5.10.0

Danct12, 2020-12-15:
```
[ 2020-12-15 13:58:28] [Renderer.cpp:104@initialize] Failed to initialize EGL
[ 2020-12-15 13:58:29] [client.cpp:48@start] Failed to start container: Failed to start container: Failed to start container
[ 2020-12-15 13:58:29] [session_manager.cpp:161@operator()] Lost connection to container manager, terminating.
[ 2020-12-15 13:58:29] [daemon.cpp:61@Run] Container is not running
[ 2020-12-15 13:58:29] [session_manager.cpp:161@operator()] Lost connection to container manager, terminating.
Segmentation fault (core dumped)
```

mogwa1, 2020-12-16:
I can confirm.  It's even coredumping here:

```Dec 16 10:25:48 pinephone anbox[4171]: Failed to start container: Failed to start container: Failed to start container
Dec 16 10:25:48 pinephone anbox[4171]: Lost connection to container manager, terminating.
Dec 16 10:25:48 pinephone anbox[4171]: Container is not running
Dec 16 10:25:48 pinephone anbox[4171]: Lost connection to container manager, terminating.
Dec 16 10:25:48 pinephone kernel: anbox0: port 1(vethT1Y6T8) entered disabled state
Dec 16 10:25:48 pinephone kernel: device vethT1Y6T8 left promiscuous mode
Dec 16 10:25:48 pinephone kernel: anbox0: port 1(vethT1Y6T8) entered disabled state
Dec 16 10:25:48 pinephone systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Dec 16 10:25:48 pinephone NetworkManager[3418]: <info>  [1608110748.4921] device (vethT1Y6T8): released from master device >
Dec 16 10:25:48 pinephone systemd[1]: Started Process Core Dump (PID 4202/UID 0).
Dec 16 10:25:56 pinephone systemd-coredump[4207]: Process 4171 (anbox) of user 1000 dumped core.
                                                  
                                                  Stack trace of thread 4177:
                                                  #0  0x0000ffff9967ebe8 aarch64_fallback_frame_state (libgcc_s.so.1 + 0xcb>
                                                  #1  0x0000ffff99680410 _Unwind_Backtrace (libgcc_s.so.1 + 0xe410)
                                                  #2  0x0000aaaaabd19f68 _ZN8backward14StackTraceImplINS_10system_tag9linux>
                                                  #3  0x0000aaaaabd1d5b4 _ZN8backward14SignalHandling12handleSignalEiP9sigi>
                                                  #4  0x0000aaaaabd1daa4 _ZN8backward14SignalHandling11sig_handlerEiP9sigin>
                                                  #5  0x0000ffff9a0457bc n/a (linux-vdso.so.1 + 0x7bc)
                                                  #6  0x0000ffff9a0457bc n/a (linux-vdso.so.1 + 0x7bc)
                                                  #7  0x0000aaaaabd5a834 n/a (anbox + 0x12a834)
                                                  #8  0x0000aaaab6b1de80 n/a (n/a + 0x0)
Dec 16 10:25:57 pinephone systemd[1]: systemd-coredump@0-4202-0.service: Succeeded.
```

beckerichmika, 2020-12-17:
Can somebody verify that the EGLDisplay parameter in the eglInitialize method is not null? Like we could print the current values of that method to see if the display is recognized.

closed 2020-12-18 by Danct12