dreemurrs-embedded/Pine64-Arch #112: Cannot boot 12/04 with F2FS (corruption issue?)

Jojonintendo, 2020-12-16:
The last release, featuring F2FS doesn't seem to boot on my PP. It is a UBports v1.2 with hardware fix to get docks to work.

The output from the serial console is this right after having flashed it to eMMC:

```
U-Boot 2020.10-3 (Dec 03 2020 - 06:43:18 +0000) Arch Linux ARM

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  2 GiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
1127 bytes read in 1 ms (1.1 MiB/s)
## Executing script at 4fc00000
gpio: pin 98 (gpio 98) value is 1
Booting from eMMC
Loading kernel...
8078747 bytes read in 347 ms (22.2 MiB/s)
Uncompressing kernel...
Uncompressed size: 18833416 = 0x11F6008
Loading initramfs...
27524685 bytes read in 1181 ms (22.2 MiB/s)
Loading dtb...
50097 bytes read in 5 ms (9.6 MiB/s)
Booting...
gpio: pin 98 (gpio 98) value is 0
Moving Image from 0x40080000 to 0x40200000, end=41460000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 485c0000, end 49fffe4d ... OK
   Loading Device Tree to 00000000485b0000, end 00000000485bf3b0 ... OK

Starting kernel ...

[    1.999758] SMCCC: SOC_ID: ARCH_SOC_ID(0) returned error: ffffffffffffffff
[    2.205433] phy id_det change to device
[   37.134313] F2FS-fs (mmcblk2p2): SIT is corrupted node# 98644 vs 92974
[   37.145081] F2FS-fs (mmcblk2p2): Failed to initialize F2FS segment manager (-117)
```

It seems to me that there is some kind of corruption on the filesystem, which is why I've tried multiple times and with different computer, but the result is always the same, it just hangs there and I get dropped to the console. Is this happening to someone else?

I'd be happy to provide any feedback necessary. Thanks in advance for your time and work!

Danct12, 2020-12-16:
May I ask what happens on the first flash and boot? 

You should be seeing a lot of kernel messages after the bootsplash hidden. Don't do anything, just keep waiting and it'll boot to the UI.

Jojonintendo, 2020-12-16:
Weirdly enough, after another try with the same procedure (boot JumpDrive, flash the eMMC, reboot), it finally booted to the UI. It might have been just bad luck, sorry for the trouble!

Danct12, 2020-12-16:
Reopened as it seems like this issue sometimes happens as well, will look into this deeper.

rprofijt, 2021-01-09:
I had the same issue with the 20201223 image, not being lazy and using conv=fsync with dd fixed it for me.

Danct12, 2021-01-24:
So I ran into the same issue while testing the image for PineTab today, turns out the card was bad (and it was brand new). I tested with another card and the OS booted fine, works properly and survived the next boot.

Danct12, 2021-06-11:
We've reverted to ext4 on newer images, there might be plan for it to return, but not now.

closed 2021-06-11 by Danct12