dreemurrs-embedded/Pine64-Arch #126: Camera not taking photos

mkzmch, 2021-01-27:
I'm running the latest image, updated just now and the camera app crashes when taking photos.  This is what I get when running it from the terminal
`Found config file at /usr/share/megapixels/config/pine64,pinephone-1.2.ini
Adding camera rear from config 
Adding camera front from config
Found postprocessor script at /usr/share/megapixels/postprocess.sh
/tmp/megapixels.gu7lVt/0.dng: Failed to allocate memory for custom tag binary object (-942346776 elements of 1 bytes each).
TIFF version 20191103
megapixels: tif_dirwrite.c:3071: TIFFWriteDirectoryTagData: Assertion 'datalength<0x80000000UL failed`

Danct12, 2021-01-27:
Any idea about this? @MartijnBraam

mkzmch, 2021-01-27:
Just tried the second most recent image, camera works fine. On the most recent image, but without pacman -Syu the issue is still there.

MartijnBraam, 2021-01-27:
@Danct12 your libtiff is broken, or megapixels needs to be rebuild against the new libtiff

Danct12, 2021-01-28:
i've rebuilt megapixels, thank you @MartijnBraam :+1:

closed 2021-01-28 by Danct12