dreemurrs-embedded/Pine64-Arch #133: Epiphany does not display any website

Barugon, 2021-02-17:
I installed Epiphany on PinePhone but it won't display any website (or the most visited pages either), just a blank window. When I navigate to a site, I can see, very briefly, the page squashed into the upper left quarter of the window and then it goes blank again.

Epiphany displays web pages just fine on other distributions.

Ocawesome101, 2021-02-20:
I haven't tried regular Epiphany in a while, but `epiphany-mobile` seems to have this issue also.

Danct12, 2021-02-20:
Maybe this is a mesa bug, I'll see if I can produce a apitrace and report it to upstream.

Danct12, 2021-02-20:
I can confirm this is a mesa bug, downgrading back to mesa 20.3.4 fixed it.

https://gitlab.freedesktop.org/mesa/mesa/-/issues/4336

Danct12, 2021-03-08:
Seems to have been fixed with latest mesa-git.

closed 2021-03-08 by Danct12