dreemurrs-embedded/Pine64-Arch #140: [Pinetab] Detached keyboard backlight default to "off"?

tomoqv, 2021-03-05:
Is it possible to set the keyboard backlight to "off" as default? As it is now, backlight is triggered every time the keyboard is attached and it doesn't automatically turn off when turning the Pinetab off.

Danct12, 2021-03-06:
I do not have the new PineTab keyboard, so can't be sure if it's controllable or not. Can you please send output of /sys/class/leds?

Might also be a duplicate of #49

tomoqv, 2021-03-06:
It is just an empty folder, or did I misunderstand you?

Danct12, 2021-03-06:
Is it with the keyboard attached? If so then I don't think it is controllable, sorry. :\

tomoqv, 2021-03-06:
input10::capslock  input10::numlock  input10::scrolllock

Danct12, 2021-04-08:
Looks like I can't do anything with it then. :(