dreemurrs-embedded/Pine64-Arch #143: Anbox with Pine64-Arch barebones (+sxmo) ?

fretzo, 2021-03-31:
Hi, i am using the pinephone with Pine64-Arch barebones and sxmo, 
and would like to try and use Anbox.
I followed the steps in the wiki, but Anbox does not launch.
Is Anbox working now on the pinephone with arch? Are there any additional steps and/or packages that need to be installed for the barebones image?
Thank you for any tips :-)

Danct12, 2021-04-01:
is the container service started? that is required for anbox to run.

fretzo, 2021-04-01:
Thanks for your answer, my bad, i thought the steps in the wiki was all it took, i have found a few tutorials, and will give them a go, 
thanks a lot for your great project !!!

Danct12, 2021-04-02:
It is, and I have made sure that those steps from the wiki is all it took.

Can you please run `sudo systemctl status anbox-container-manager`?

fretzo, 2021-04-02:
Hi, it says it is active and running (in green).
When i then try to launch anbox (from the sxmo apps launcher menu) nothing happens.
To be clear, I only did these steps:
$ sudo pacman -Syu
$ sudo pacman -S anbox-image-aarch64

Raezroth, 2021-04-02:
> Hi, it says it is active and running (in green).
> When i then try to launch anbox (from the sxmo apps launcher menu) nothing happens.
> To be clear, I only did these steps:
> $ sudo pacman -Syu
> $ sudo pacman -S anbox-image-aarch64

Have you tried running anbox from terminal while specifying the GDK_BACKEND? ( Example: $ GDK_BACKEND=x11 anbox %f ) I will install sxmo (today) over the bare bones image and see about testing it myself for more support

fretzo, 2021-04-02:
It now works. When launching anbox from the terminal, i get an error message, and a longer command to use to launch anbox. Using this command, launches Anbox:

`anbox launch --package=org.anbox.appmgr --component=org.anbox.appmgr.AppViewActivity`

Sorry for the hassle. 
Btw, i use this installer for sxmo: https://github.com/justinesmithies/sxmo-alarm

closed 2021-04-03 by Danct12