dreemurrs-embedded/Pine64-Arch #145: Plasma Mobile - Auto Rotate Not Working

AngelTomkins, 2021-04-04:
**Problem**
Auto rotate started off working, but after using plasma mobile for a bit, the auto screen rotation no longer works. When starting the phone up, it stats with auto rotate disabled by default(I believe when it was working it was starting enabled). When I try to enable the auto rotate in the pull down menu, it shows its on afterwards, but still doesn't work, and I cannot turn off the visual indicator on the drop down menu.

See here:
https://user-images.githubusercontent.com/67720650/113523158-1a01c280-9574-11eb-85be-b59a4318c606.mp4

**What I have done**
I installed plasma `system settings` and used that to try to enable the auto rotate, but that did not work either, though manual rotation through that settings program does work. I checked to make sure that iio-sensor-proxy was running, and it is. I have restarted multiple times and nothing changes. Some times it lets me turn on and off the visual indicator on the drop down, but still the auto rotate doesn't work.

AngelTomkins, 2021-04-14:
Output of ` kscreen-doctor -j` with the autorotation set as on in the top dropdown menu

```
{
    "features": 54,
    "outputs": [
        {
            "clones": [
            ],
            "connected": true,
            "currentModeId": "0",
            "enabled": true,
            "followPreferredMode": false,
            "icon": "",
            "id": 1,
            "modes": [
                {
                    "id": "0",
                    "name": "720x1440@60",
                    "refreshRate": 60.00600051879883,
                    "size": {
                        "height": 1440,
                        "width": 720
                    }
                }
            ],
            "name": "unknown DSI-1-unknown",
            "pos": {
                "x": 0,
                "y": 0
            },
            "preferredModes": [
                "0"
            ],
            "primary": true,
            "replicationSource": 0,
            "rotation": 1,
            "scale": 2,
            "size": {
                "height": 1440,
                "width": 720
            },
            "sizeMM": {
                "height": 136,
                "width": 68
            },
            "type": 7
        }
    ],
    "screen": {
        "currentSize": {
            "height": 720,
            "width": 360
        },
        "id": 0,
        "maxActiveOutputsCount": 1,
        "maxSize": {
            "height": 64000,
            "width": 64000
        },
        "minSize": {
            "height": 0,
            "width": 0
        }
    },
    "tabletModeAvailable": true,
    "tabletModeEngaged": true
}
```

AngelTomkins, 2021-04-14:
The issue seems to be an upstream issue. The solution was to remove the ~/.local/share/kscreen directory. For those who would wish to study it, the directory is linked in zip format.
[kscreen.zip](https://github.com/dreemurrs-embedded/Pine64-Arch/files/6313128/kscreen.zip)

I believe the issue relates to having an external monitor used, since there are a lot of entries including the name of my external monitor that I have used with the pinephone beforehand.

AngelTomkins, 2021-04-14:
After some more testing, actually the solution is having the device boot up with wifi disabled, I believe this could have to do with the integration with the plasma password management for wifi networks.

AngelTomkins, 2021-04-15:
After some more testing, it does not matter if the device auto connects to a network, it does not matter what network it is, it does not matter if the modem kill switch is on/off, it seems to only matter if the wifi is switched on at startup. Deleting and working around the kdewallet integration showed no changes to the behaviour.

Danct12, 2021-06-11:
Can't reproduce, is it still a issue?

closed 2021-06-14 by Danct12