dreemurrs-embedded/Pine64-Arch #158: Modem not working after modemmanager upgrade

dariox86, 2021-05-05:
Modem does not work after upgrading modemmanager from version 1.16.2-1 to 1.16.4-1.

Danct12, 2021-05-05:
Mind describing the symptoms? This is not really helpful.

dariox86, 2021-05-05:
Sorry about that. I thought I was not the only one experiencing the issue. Wi-Fi and mobile network do not work after upgrading modemmanager. I think the modem does not initialize at all. I have no mobile nor Wi-Fi icons in Phosh. Does it work for you? What logs could be helpful?

DrRac27, 2021-05-05:
Does it work if you downgrade modemmanager? You can do so using something like `sudo pacman -U /var/cache/pacman/pkg/modemmanager....`.
I had a similar problem and for me the problem was not modemmanager but that `/boot` was not mounted while I updated so the problem was that after an update the new kernel was not booted and the old kernel had no modules anymore. You can find out if that's your problem as well if you check if your running kernel fits your installed modules. `uname -a` prints the version of your running kernel, `ls /lib/modules` shows the version of the installed modules (or `pacman -Qi linux-pine64`).
If that is your problem as well you can check if `/boot` is mounted using `mount | grep boot` and if it is not mounted you can run `sudo rm -r /boot/* && sudo mount /dev/mmcblk2p1 /boot && sudo pacman -S linux-pine64`. This can break your system if you do something wrong and I wrote this tired and out of my head and I didn't verify anything so you should better check the commands first...
(@Danct12: ) I don't know if this is a problem everybody has because I played quite a bit with my install so maybe I messed up my fstab or something.

dariox86, 2021-05-06:
@DrRac27, I appreciate your feedback. Yes, it does work if I downgrade modemmanager. In fact that is what I did to keep using my PinePhone. Later I will check the rest to troubleshoot the issue.

Danct12, 2021-05-06:
Have you tried to update to the latest version again? this should fix your problem.

Regarding Wi-Fi, that's actually not controlled by ModemManager so that shouldn't be affected.

dariox86, 2021-05-06:
Updated to latest version. It is working again. Thank you.

closed 2021-05-07 by Danct12

Brueggus, 2021-05-12:
I'm having the same issue after updating to modemmanager 1.16.4-1:

```
[alarm@piphone ~]$ journalctl -k -g modem
-- Journal begins at Fri 2021-04-16 15:12:03 UTC, ends at Wed 2021-05-12 10:15:54 UTC. --
Mai 12 10:05:40 piphone kernel: cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
Mai 12 10:05:59 piphone kernel: usbserial: USB Serial support registered for GSM modem (1-port)
Mai 12 10:05:59 piphone kernel: option 2-1:1.0: GSM modem (1-port) converter detected
Mai 12 10:05:59 piphone kernel: usb 2-1: GSM modem (1-port) converter now attached to ttyUSB0
Mai 12 10:05:59 piphone kernel: option 2-1:1.1: GSM modem (1-port) converter detected
Mai 12 10:05:59 piphone kernel: usb 2-1: GSM modem (1-port) converter now attached to ttyUSB1
Mai 12 10:05:59 piphone kernel: option 2-1:1.2: GSM modem (1-port) converter detected
Mai 12 10:05:59 piphone kernel: usb 2-1: GSM modem (1-port) converter now attached to ttyUSB2
Mai 12 10:05:59 piphone kernel: option 2-1:1.3: GSM modem (1-port) converter detected
Mai 12 10:05:59 piphone kernel: usb 2-1: GSM modem (1-port) converter now attached to ttyUSB3
Mai 12 10:06:40 piphone kernel: option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
Mai 12 10:06:40 piphone kernel: option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
Mai 12 10:06:40 piphone kernel: option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
Mai 12 10:06:40 piphone kernel: option1 ttyUSB3: GSM modem (1-port) converter now disconnected from ttyUSB3
```

```
[alarm@piphone ~]$ sudo journalctl -u ModemManager
=> see attachment
```
[modemmanager.log](https://github.com/dreemurrs-embedded/Pine64-Arch/files/6465477/modemmanager.log)

> Have you tried to update to the latest version again? this should fix your problem.

I tried removing and reinstalling modemmanager 1.16.4-1 without success.