dreemurrs-embedded/Pine64-Arch #170: WPA Enterprise wifi login not fitting to scale, and in recent update, not connecting

EnderNightLord-ChromeBook, 2021-05-30:
In the recent update I got a "Authentication ype of wifi network "the wifi network name" not supported when trying to connect to my schools network, as my other phone is dead and I'm daily driving my pinephone (it is pretty great btw). 

I don't know what would be causing the problem as it was working fine before. The other issue is that the screen to put in the details for wpa enterprise networks is overflowing in many places, I'd like to see it fixed if possible.

bfra2373, 2022-11-29:
[phosh-mobile-settings](https://gitlab.gnome.org/guidog/phosh-mobile-settings) might be of use (it forces scale down on pages that won't). I'll try tomorrow and let you know if this work's because I have the exact same issue.

bfra2373, 2022-11-29:
I did test this workaround today and it work just fine.