dreemurrs-embedded/Pine64-Arch #171: Discord Link is invalid

RobertFULL30, 2021-06-10:
This (phosh) is the best I've found for the Pinephone and I'd like to be able to contribute 

The discord link on https://danctnix.org is expired

Robert

Danct12, 2021-06-10:
I have already removed the invalid link from the repository, but not on the website. Currently, there's no plan to reopen the Discord server yet.

closed 2021-06-10 by Danct12

Danct12, 2021-06-10:
No plan to reopen the Discord server for now.

RobertFULL30, 2021-06-10:
How about installing a version of discourse on https://danctnix.org 

I'm available to help, this build is the best I've found available and could use a more viable discussion platform IMO

There are still bugs but I can at least make calls and want to encourage and promote this.

Danct12, 2021-06-10:
https://forum.pine64.org/forumdisplay.php?fid=159

There is also Matrix channel (#archmobile:kde.org) and Telegram (@archmobile) as replacement for Discord.