dreemurrs-embedded/Pine64-Arch #175: Slow GTK4 apps since GSK_RENDERER change.

caveman250, 2021-06-15:
i noticed today after installing the newest image that performance had severely degraded in most of my gtk4 apps, especially when scrolling.

there seem to be certain css properties that make the problem more pronounced such as borders, but even without these there is a noticible difference.

running apps with GSK_RENDERER=cairo completly fixes the problem.

lolgzs, 2021-07-10:
Hi, 

On manjaro-arm Megapixels (GTK4) video preview  was slow. I found this issue then changed /etc/profile.d/gsk-renderer-gl.sh 
`export GSK_RENDERER=ngl `
(instead of gl)

And now the video renders smoothly.

bridadan, 2021-07-15:
Just heads up, the reason why the `GSK_RENDERER` was reverted to `gl` is detailed here: https://gitlab.gnome.org/GNOME/gtk/-/issues/3894

A fix has been merged: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/11872

Danct12, 2021-08-05:
Fixed.

https://github.com/dreemurrs-embedded/Pine64-Arch/commit/99df0d2cc04e64d4c46e12d783328d996d23afa0 https://github.com/dreemurrs-embedded/Pine64-Arch/commit/ae1537a71fd29ec4a5f5a000071eebd49dc3ce27

closed 2021-08-05 by Danct12