dreemurrs-embedded/Pine64-Arch #176: Anbox does not start (yet) with latest versions of lxc and boost-libs

dariox86, 2021-06-16:
Same as closed issue #173. Nothing has changed so far. lxc and boost-libs must be kept at version 1:4.0.6-1 and 1.75.0-5 respectively for Anbox to work.

Danct12, 2021-06-16:
Works perfectly fine for me on anbox-0_git20210320-2, lxc-4.0.6-1 and boost-libs 1.76.0-1

```
[alarm@danctnix ~]$ pacman -Ss anbox
danctnix/anbox 0_git20210320-2 [installed]

[alarm@danctnix ~]$ pacman -Ss boost-libs
extra/boost-libs 1.76.0-1 [installed]
    Free peer-reviewed portable C++ source libraries (runtime libraries)

[alarm@danctnix ~]$ lxc-info --version
4.0.6

[alarm@danctnix ~]$ ps aux | grep surfaceflinger
alarm       4240  0.3  1.1  56124 23568 pts/0    S<l  09:27   0:04 /system/bin/surfaceflinger
alarm       5481  0.0  0.0   5992  1760 pts/1    S+   09:50   0:00 grep surfaceflinger
[alarm@danctnix ~]$ 
```

dariox86, 2021-06-16:
The problem is that I see no anbox-0_git20210320-2 in the repositories. I noticed you have updated the PKGBUILD for Anbox on GitHub, but I don't see it coming when I update with pacman -Syu. Actually I see no Anbox at all when doing «pacman -S anbox». It says package was not found. It looks like I am missing the danctnix repo.

dariox86, 2021-06-17:
I don't know why I missed a repository. I needed to add this to /etc/pacman.conf:
```
[danctnix]
SigLevel = Optional TrustAll
Server = https://p64.arikawa-hi.me/$repo/$arch
```
Is this the proper way to add your custom repository?

Anbox works albeit with older lxc-1:4.0.6-1.

Danct12, 2021-06-18:
yup, add it above the [core] repository.

dariox86, 2021-06-18:
How come it disappeared? Was it my mistake?

dariox86, 2021-07-29:
Anbox works with latest versions of lxc and boost-libs. I think the issue can be closed.

closed 2021-07-29 by Danct12