dreemurrs-embedded/Pine64-Arch #180: 2021-06-13 Phosh - Booting Stall

cepsbow, 2021-06-20:
Through repeated flashes this issue has persisted. The booting process hangs at:

"WARNING: Unimplemented Standard Service Call: 0x84000050"

This occurs directly after loading the kernel image into the ramdisk. I have not been able to get any response from the phone once it enters this state, aside from directly powering it off via button or battery. Barebones and plasma has not show this issue for me whatsoever.

lazy-1, 2021-06-20:
I've fresh installed and updated archlinux-pinephone-phosh-20210613.img.xz more than 3 dozen times since it has been created. Not once have I had a boot freeze. This is on 16 cards. Different sizes, makes and models. 
Please specify the command you use to flash the image and what process you follow. for me I do the following 
unxz -c  archlinux-pinephone-phosh-20210613.img.xz | dd of=/dev/sde bs=1M status=progress && sync
then my script waits 2 mins just on the safe side before it mounts it and copies my default stuff into /home/alarm along with /etc files I use. waits another min for safe finish of writes and umounts the 4.3G partition.

cepsbow, 2021-06-20:
I'm flashing the emmc directly using Etcher

lazy-1, 2021-06-20:
I can't comment on emmc mass installs. I've only done that about 3 times since that install version and each has been without issue but know 3 is not much of a statistic. 
Any more info I am sure will help track the problem. What version Pinephone? Is it before or after you update? and anything else you can think of.

cepsbow, 2021-06-20:
The previous release from May worked fine, I'm running a Pinephone 1.2b hardware with 3GB and 32GB EMMC. The EMMC just bugged out on me just now and wouldn't mount for a few boots. Might have to break this one down and examine the hardware.

cepsbow, 2021-06-20:
Used dd if=/dev/null of=/dev/sdb first and it loaded finally. Probably should clear the emmc or add a sanity check to the boot process, i'll see if this is truly isolated though. I'm gonna debug my emmc some and make sure its performing up to par, may be just a junk chip.

closed 2021-06-20 by cepsbow