dreemurrs-embedded/Pine64-Arch #182: [Phosh] .desktop with "Terminal=True": xterm not found

langfingaz, 2021-06-24:
## Actual behavior

When I start a desktop entry from the application drawer that has `Terminal=true` in it's `.desktop` file (e.g. `Htop`), then this application does not show up.

When I start the same application with

```shell
gtk-launch htop.desktop
```

then I get the following error:

```
sh: line 1: exec: xterm not found
```

## Expected behavior

The installed terminal, in my case [KGX](https://gitlab.gnome.org/ZanderBrown/kgx) should open up and the application should be running in there.

[[image]](https://source.puri.sm/Librem5/phoc/uploads/042951f6f73f448c9482ec7d984c2dd9/2021-06-24-103502.png)

## My setup

The above behavior occurs on my PinePhone with DanctNIX Arch Linuxarm and **Phosh** as the desktop environment.

I think **Plasma Mobile** is not affected as another terminal is used there.

## Possible solution

I had first submitted this issue report at [source.puri.sm](https://source.puri.sm/Librem5/phoc/-/issues/219#note_161472) and they helped me out a bit:

- The problem is that as of now KGX is not included in [gdesktopappinfo.c](https://gitlab.gnome.org/GNOME/glib/-/blob/main/gio/gdesktopappinfo.c#L2581) of glib2. But the "the termapp problem is well known and proper solutions are being worked on".
- A quick fix for me was archieved with `sudo ln -s /usr/bin/kgx /usr/bin/konsole`. Maybe this fix could be included in your [PKGBUILD for KGX](https://github.com/dreemurrs-embedded/Pine64-Arch/blob/master/PKGBUILDS/phosh/kgx/PKGBUILD)? One would just need to add this line to `package(){}`:

```shell
  ln -sf /usr/bin/kgx "$pkgdir"/usr/bin/konsole
```

---

What do you think? Should I create a pull request with the proposed changes?

Danct12, 2021-06-24:
> I think Plasma Mobile is not affected as another terminal is used there.

It is affected as Plasma Mobile image uses qmlkonsole which technically is **not** konsole.

> The problem is that as of now KGX is not included in gdesktopappinfo.c of glib2. But the "the termapp problem is well known and proper solutions are being worked on".

We can fork glib2 and make the change, the issue is that we need to keep up with upstream and the rebase process is not automated. During that moment, something *might* break.

> A quick fix for me was archieved with sudo ln -s /usr/bin/kgx /usr/bin/konsole. Maybe this fix could be included in your PKGBUILD for KGX?

**No.** Doing that will create a file conflict for the actual Konsole package and what if they use some sort of hybrid plasma + phosh setup?

langfingaz, 2021-06-24:
The problem is that `kgx` is not detected in `gdesktopappinfo.c`. But creating a symlink in `/usr/bin` to any of these terminal emulators should work:

```c
          if (check == NULL)
            check = g_find_program_in_path ("tilix");
          if (check == NULL)
            check = g_find_program_in_path ("konsole");
          if (check == NULL)
            check = g_find_program_in_path ("nxterm");
          if (check == NULL)
            check = g_find_program_in_path ("color-xterm");
          if (check == NULL)
            check = g_find_program_in_path ("rxvt");
          if (check == NULL)
            check = g_find_program_in_path ("dtterm");
```

I have just randomly picked `konsole` and it worked for me on Phosh.

But I agree: This symlink hack is not nice as it does not work when `konsole` is installed.

A glib2 fork seems like some work. Maybe we should check out how this is done by Manjaro for PinePhone?

Or we could pick one of the above terminal emulators which is more unlikely to be used (e.g. `/usr/bin/ddterm`) and add a `conflicts=(ddterm)` to the PKGBUILD. This does however not solve the problem completely but might be better than using `konsole`.

Danct12, 2021-06-26:
I have checked [glib2](https://gitlab.manjaro.org/packages/core/glib2) from Manjaro side, it's all unchanged.

But make a symlink to dtterm as a workaround seems to be a good idea as we don't want to manually rebase glib2 every time there's a new update.

It's also a [**very ancient** terminal emulator](http://cdesktopenv.sourceforge.net/man1/dtterm.html) that nobody uses. Even if they want a CDE on Arch setup, they wouldn't be doing it on the Phosh image, but rather on barebone.

closed 2023-08-29 by Danct12