dreemurrs-embedded/Pine64-Arch #184: Chatty crash on start

dariox86, 2021-06-24:
Chatty crash with this message in the terminal:
`chatty: symbol lookup error: chatty: undefined symbol: _ZN4i18n12phonenumbers11PhoneNumberC1EPN6google8protobuf5ArenaE`

This happens when the following packages are installed:
`protobuf-3.17.3-1
protobuf-c-1.4.0-1
libphonenumber-8.12.25-1
evolution-data-server-3.40.2-2`

Downgrading to the following versions fixes the issue:
`protobuf-3.16.0-1
protobuf-c-1.3.3-4
libphonenumber-8.12.24-1
evolution-data-server-3.40.2-1`

Danct12, 2021-06-26:
That's odd, this is the output after I updated my force updated the repository on my phone.
```
[alarm@danctnix ~]$ sudo pacman -Syy
:: Synchronizing package databases...
 pine64                                             3.1 KiB  2.96 KiB/s 00:01 [############################################] 100%
 plasma                                             4.6 KiB  4.37 KiB/s 00:01 [############################################] 100%
 phosh                                              6.8 KiB  10.4 KiB/s 00:01 [############################################] 100%
 danctnix                                           9.0 KiB  8.59 KiB/s 00:01 [############################################] 100%
 core                                             198.3 KiB   166 KiB/s 00:01 [############################################] 100%
 extra                                              2.3 MiB   704 KiB/s 00:03 [############################################] 100%
 community                                          5.3 MiB  1059 KiB/s 00:05 [############################################] 100%
 alarm                                             55.5 KiB  59.6 KiB/s 00:01 [############################################] 100%
 aur                                                6.0 KiB  8.01 KiB/s 00:01 [############################################] 100%
[alarm@danctnix ~]$ sudo pacman -Ss chatty
phosh/chatty 0.3.1-2 [installed]
    Messaging client
[alarm@danctnix ~]$ pacman -Ss anbox
danctnix/anbox 0_git20210320-3 [installed]
    
danctnix/anbox-image-aarch64 0_20200520-6 [installed]
    Anbox image for ARM64
[alarm@danctnix ~]$
```

dariox86, 2021-06-26:
Weird. I tried once again with the latest versions. It is not working for me. I get the error message reported above when launching the application from the terminal. The problem lies with either protobuf-c or protobuf. The other packages have to be downgraded only as dependencies of the older versions of these two packages.

dariox86, 2021-07-13:
Sorry, I was missing all the repositories above [danctnix]. [pine64], [plasma] and [phosh] were not present in pacman.conf. I don't know why. Chatty works.

closed 2021-07-13 by dariox86