dreemurrs-embedded/Pine64-Arch #189: [Phosh] Mobile Data defaults to on

fdservices, 2021-07-02:
Whenever I restart the pinephone, mobile data is set to on. 
Steps to reproduce:
  Set Settings > Mobile > Mobile Data to Off
  Restart pinephone

If I knew where the settings are stored then I could always reset it at start up, but I am still learning :)

fdservices, 2021-07-12:
I worked it out. The commands to turn the modem data on and off are:
`$ sudo nmcli d connect cdc-wdm0`
`$ sudo nmcli d disconnect cdc-wdm0`
Verify the result with
`$ nmcli d`
The Settings programme needs to be restarted in order to show the changes

fdservices, 2021-07-14:
Sorry, I closed this because I found a fix for the issue. The problem remains.
The problem is that if Mobile Data is manually set to off, then after a restart that setting is not respected and Mobile Data is set back to on.
This does not happen in other operating systems.

Schmiddiii, 2025-02-28:
I think I figured out how to do it:

```sh
nmcli connection
```

This should list all connections your devices has.
There should be one with type `gsm` (in my case I had two for some reason).
Copy its UUID.

```sh
nmcli connection show <uuid>
```

This should list properties of the connection.
In particular, `connection.autoconnect` should be `yes`.
You can modify that using:

```sh
nmcli connection modify <uuid> connection.autoconnect no
```

See also [this StackExchange answer](https://unix.stackexchange.com/questions/683293/how-to-set-connection-to-down-by-default-with-nmcli).

Ideally, whether a connection is activated or not should be persisted between suspends and reboots.
But for me personally, being deactivated is the better default (I mostly use WiFi).