dreemurrs-embedded/Pine64-Arch #191: Display rotation working erratically with iio-sensor-proxy-3.1-1

dariox86, 2021-07-11:
When automatic screen rotation is enabled with iio-sensor-proxy-3.1-1, the display is shown in landscape mode when PinePhone is held vertically and is shown in portrait mode while it is held horizontally. It spins in all the wrong directions, even upside down. I can't get it right unless I disable automatic rotation. Downgrading to iio-sensor-proxy-3.0-1 fixes the issue.

Danct12, 2021-07-12:
I'm currently on iio-sensor-proxy-3.1-1 and rotation is fine. Would you mind running monitor-sensor and specify which direction goes in the log?

dariox86, 2021-07-12:
Real: normal. Reported: normal shortly followed by left-up, randomly switching between the two multiple times.
Real: right-up. Reported: right-up, normal, left-up, normal... indefinitely without moving the smartphone.
Real: upside-down. Reported: right-up, normal, right-up, normal... indefinitely.
Real: left-up. Reported: left-up shortly followed by bottom-up and sometimes randomly switching between the two.

closed 2024-08-29 by Danct12