dreemurrs-embedded/Pine64-Arch #194: Memory card is not detected

SAPllak, 2021-07-18:
I used jumpdrive and the memorycard works exept when I have a memory card and I have booted the system it doesn't detect my music from it.

Danct12, 2021-07-20:
Right now we don't automatically mount the card, check if /dev/mmcblk0(pX) exists, if so, mount it.

Melechtna, 2021-09-01:
I personally just added it to FStab as my homefolder, mounts on its own that way. Not sure if that's what you're looking for.

lafleurdeboum, 2021-12-07:
Installing `gvfs` solved this issue for me I think - my ext4-formatted SD card gets mounted at boot, and I can see in the mount options it was mounted by udisks2.

That's probably unrelated, but my pinephone refuses to boot if said SD card is inserted at boottime, even though I followed re-use instructions at [pine's wiki](https://wiki.pine64.org/wiki/PinePhone_Installation_Instructions#Reuse_SD_card_for_data_storage_on_system_booting_from_eMMC) (twice).