dreemurrs-embedded/Pine64-Arch #196: DP/HDMI doesn't work with PP Beta convergence

Ferenc-, 2021-07-24:
This is a freshly received device,
the image is `archlinux-pinephone-phosh-20210713.img.xz`,  I tried multiple monitors and docking stations.
All the docking stations and the monitors are otherwise functional together.

The filtered `dmesg` after docking:
```
[ 3279.533970] anx7688 1-0028: plug irq (cd=1)
[ 3279.534269] anx7688 1-0028: plug irq (cd=0)
[ 3279.534793] anx7688 1-0028: plug irq (cd=1)
[ 3279.534993] anx7688 1-0028: plug irq (cd=0)
[ 3279.536781] anx7688 1-0028: plug irq (cd=1)
[ 3279.536979] anx7688 1-0028: plug irq (cd=1)
[ 3279.537038] anx7688 1-0028: plug irq (cd=1)
[ 3279.537347] anx7688 1-0028: plug irq (cd=0)
[ 3279.538275] anx7688 1-0028: plug irq (cd=1)
[ 3279.538536] anx7688 1-0028: plug irq (cd=0)
[ 3279.539876] anx7688 1-0028: plug irq (cd=1)
[ 3279.539968] anx7688 1-0028: plug irq (cd=0)
[ 3279.540246] anx7688 1-0028: plug irq (cd=1)
[ 3279.540380] anx7688 1-0028: plug irq (cd=0)
[ 3279.540488] anx7688 1-0028: plug irq (cd=1)
[ 3279.547475] anx7688 1-0028: cable inserted
[ 3279.587262] anx7688 1-0028: power enabled
[ 3279.651569] anx7688 1-0028: eeprom0 = 0x03
[ 3279.651650] anx7688 1-0028: fw loaded after 40 ms
[ 3279.651886] anx7688 1-0028: OCM firmware loaded (version 0x2312)
[ 3279.652886] anx7688 1-0028: send pd packet cmd=0x00 05 00 32 90 01 26 12
[ 3279.701769] anx7688 1-0028: send pd packet cmd=0x01 05 01 2c 91 01 26 16
[ 3279.702347] anx7688 1-0028: send pd packet cmd=0x02 11 02 00 00 00 ec 00 00 00 00 00 00 00 00 39 00 00 51 77
[ 3279.703228] anx7688 1-0028: send pd packet cmd=0x03 05 03 00 00 01 ff f8
[ 3279.703767] anx7688 1-0028: OCM configuration completed
[ 3279.704782] anx7688 1-0028: status changed to 0x00
[ 3279.704832] anx7688 1-0028: cc_status changed to CC1 = SRC.Open CC2 = SRC.Open
[ 3279.704848] anx7688 1-0028: DP state changed to 0x00
[ 3279.704861] anx7688 1-0028: VCONN role change to SINK
[ 3279.704910] anx7688 1-0028: DATA role change requested to UFP
[ 3280.150753] anx7688 1-0028: cc_status changed to CC1 = SNK.Power3.0 CC2 = SRC.Open
[ 3280.182915] anx7688 1-0028: recv ocm message cmd=0xf2 01 f2 0d
[ 3280.182984] anx7688 1-0028: received HARD_RST, idiot firmware is bored
[ 3281.107327] anx7688 1-0028: recv ocm message cmd=0x00 05 00 c8 90 01 2e 74
[ 3281.107410] anx7688 1-0028: received SRC_CAP
[ 3281.107421] anx7688 1-0028: SRC_CAP PDO_FIXED (5000mV 2000mA)
[ 3281.107705] anx7688 1-0028: RDO max voltage = 5000mV, max power = 10000mW, PD current limit = 2000mA
[ 3281.110214] anx7688 1-0028: recv ocm message cmd=0xf0 03 f0 16 00 f7
[ 3281.110232] anx7688 1-0028: received response to PWR_OBJ_REQ (SUCCESS)
[ 3281.683256] anx7688 1-0028: updating power mode to PD, current limit 2000mA (0 => BC1.2)
[ 3281.683333] anx7688 1-0028: disabling USB BC 1.2 detection
[ 3281.683455] anx7688 1-0028: setting vbus_in current limit to 2000 mA
[ 3281.683571] anx7688 1-0028: enabling vbus_in power path
[ 3281.684361] anx7688 1-0028: BC 1.2 result: SDP
[ 3281.721718] anx7688 1-0028: recv ocm message cmd=0x11 01 11 ee
[ 3281.721787] anx7688 1-0028: received DSWAP_REQ
[ 3281.722214] anx7688 1-0028: status changed to 0x20
[ 3281.722518] anx7688 1-0028: DATA role change requested to DFP
[ 3281.776174] anx7688 1-0028: recv ocm message cmd=0xf0 03 f0 13 00 fa
[ 3281.776253] anx7688 1-0028: received response to VCONN_SWAP_REQ (SUCCESS)
[ 3281.776626] anx7688 1-0028: status changed to 0x24
[ 3281.776638] anx7688 1-0028: VCONN role change to SOURCE
[ 3281.800851] anx7688 1-0028: recv ocm message cmd=0x02 11 02 39 06 00 6c 00 00 00 00 00 01 00 00 d3 00 00 11 5d
[ 3281.800917] anx7688 1-0028: received DP_SNK_IDENTITY
[ 3281.804329] anx7688 1-0028: recv ocm message cmd=0x03 05 03 00 00 01 ff f8
[ 3281.804404] anx7688 1-0028: received SVID
[ 3281.808389] anx7688 1-0028: recv ocm message cmd=0x27 09 27 03 80 01 ff 05 0c 00 00 3c
[ 3281.808449] anx7688 1-0028: received unknown message 0x27
[ 3281.818932] anx7688 1-0028: recv ocm message cmd=0x19 01 19 e6
[ 3281.818999] anx7688 1-0028: received DP_ALT_ENTER
[ 3283.730685] anx7688 1-0028: DP state changed to 0x03
```

`wlr-randr` only picks up the built-in one:
```
DSI-1 "<Unknown> <Unknown>  (DSI-1)"
  Physical size: 68x136 mm
  Enabled: no
  Modes:
    720x1440 px, 60.006001 Hz (preferred)

```
But not the HDMI:
```
wlr-randr --output card1-HDMI-A-1 --on
unknown output card1-HDMI-A-1
```

The sysfs  is there:
```
ls -lah /sys/devices/platform/display-engine/drm/card1/card1-HDMI-A-1/
total 0
drwxr-xr-x 3 root root    0 Jan  1  1970 .
drwxr-xr-x 5 root root    0 Jan  1  1970 ..
lrwxrwxrwx 1 root root    0 Jul 24 15:04 ddc -> ../../../../soc/1ee0000.hdmi/i2c-0
lrwxrwxrwx 1 root root    0 Jul 24 15:04 device -> ../../card1
-r--r--r-- 1 root root 4.0K Jul 24 15:04 dpms
-r--r--r-- 1 root root    0 Jul 24 15:04 edid
-r--r--r-- 1 root root 4.0K Jul 13 00:00 enabled
-r--r--r-- 1 root root 4.0K Jul 24 15:04 modes
drwxr-xr-x 2 root root    0 Jul 13 00:00 power
-rw-r--r-- 1 root root 4.0K Jul 24 15:04 status
lrwxrwxrwx 1 root root    0 Jan  1  1970 subsystem -> ../../../../../../class/drm
-rw-r--r-- 1 root root 4.0K Jan  1  1970 uevent
```

Any ideas?

Danct12, 2021-07-27:
Known issue, seems to mostly affect Beta Edition. I recommend get in contact with @megous and try to debug the issue.

Hints on debugging HDMI output on PinePhone: https://xnux.eu/log/#039

Ferenc-, 2021-07-28:
Hi @Danct12,
Thanks for the info. I read the blog post from @megous, and checked a few things.
So the dri state is:

```
[root@danctnix alarm]# grep -A3 HDMI-A-1  /sys/kernel/debug/dri/1/state 
connector[52]: HDMI-A-1
        crtc=(null)
        self_refresh_aware=0
```
Which according to the blog post, means that.

> so if you see connector[58]: HDMI-A-1 crtc=(null) [...] it means that your compositor simply did not enable the HDMI output, so your monitor stays off. If this happens on Xorg, xrandr --auto is the quickest way to at least get some output.

So that means, that `wlr-randr` should just work too?  But unfortunately it doesn't:

```
[alarm@danctnix ~]$ wlr-randr --output HDMI-A-1 --on 
unknown output HDMI-A-1
```
Or am I doing something wrong?
In any case, I have checked this with megi's multi image with sxmo's xorg, and there `xrandr --auto` didn't help either,
so it looks like it's not just one compositor that can't deal with this HDMI output.

Then next step:
> If /sys/kernel/debug/dri/*/state configuration looks correct, but your monitor is still not on, it's either a kernel or perhaps a HW issue.

The config looks OK to me, but correct me if I am wrong. 

Next:
> you should check if you see some received SRC_CAP messages or received SVID or DP_ALT_ENTER messages in your dmesg after the last cable plugin event.

I get those, so not a hardware issue after all?
Does it mean that only the kernel is left to blame?

Ferenc-, 2021-09-04:
megi's blogpost about the issue: https://xnux.eu/log/#045

Pinienzapfen, 2021-11-17:
Works for me now with kernel pine64/linux-megi-5.15.2-1.

Ferenc-, 2021-12-25:
Works with current kernels, closing the issue.

closed 2021-12-25 by Ferenc-