dreemurrs-embedded/Pine64-Arch #199: Anbox on Plasma Mobile

AngelTomkins, 2021-08-03:
I remember there being a solution on how to get Anbox working under Plasma Mobile, which had to do with changing the back-end in the .desktop file. I lost that and not sure how to get it. I think it would be a good idea to add in the wiki page as well the solution for plasma mobile using Anbox.

Could someone share the command to change the back-end properly? Thanks!!

AngelTomkins, 2021-08-04:
I was able to find the command that needs to precede launching anbox, the full command is here:
`EGL_PLATFORM=x11 anbox launch --package=org.anbox.appmgr --component=org.anbox.appmgr.AppViewActivity`

I believe this should go in the wiki for anbox for plasma users. Do note, that this does not work under phosh

Danct12, 2021-09-19:
Anbox will probably never get Wayland support anytime soon, so I decided to modify the environment variable: 2488720b83595d4bbaf50829a506bcd8c68e25be

closed 2021-09-19 by Danct12