dreemurrs-embedded/Pine64-Arch #203: Both wired & wireless internet disappeared after upgrade to kernel 5.12.19 (barebones).

latgarf, 2021-08-09:
Others see this too?

Danct12, 2021-08-10:
Can you check lsmod and see if modules are loaded?

latgarf, 2021-08-10:
sun4i_lradc_keys  is the only loaded module.

Danct12, 2021-08-10:
That's weird, can you try `modprobe 8723cs` and see if it loads the wifi module? You can also try to reinstall the kernel if that doesn't help.

latgarf, 2021-08-10:
```
# modprobe 8723cs
modprobe: FATAL: Module 8723cs not found in directory /lib/modules/5.12.19-1-danctnix
```

Danct12, 2021-08-10:
Huh, that seems like the kernel hasn't fully updated, can you check if /boot is mounted on your device? You may have a broken fstab.

latgarf, 2021-08-10:
`/dev/mmcblk0p2  is mounted at  /boot`

Danct12, 2021-08-10:
What's in it? Usually we keep rootfs at p2 and boot at p1. Also, can you post the output of /etc/fstab?

latgarf, 2021-08-10:
```
# cat /etc/fstab
# /dev/loop0p2
UUID=...   / ext4  rw,relatime  0 1

# /dev/loop0p1
UUID=...   /boot vfat rw,relatime...,errors=remount-ro  0 2
```

Danct12, 2021-08-10:
Try to reinstall your kernel then (pacman -S linux-pine64) and reboot, see if that fixes it.

latgarf, 2021-08-10:
What's the command to reinstall kernel, please?
Is this expected to work without internet connection?

Danct12, 2021-08-10:
Download this and transfer it to your device https://p64.arikawa-hi.me/pine64/aarch64/linux-pine64-5.13.8-1-aarch64.pkg.tar.xz

Then run `pacman -U linux-pine64-5.13.8-1-aarch64.pkg.tar.xz`

latgarf, 2021-08-10:
Installed, rebooted - now there's wifi :)  Thanks!  I'll try to upgrade now pacman -Syu.  I guess the initial problem was because of power failure during kernel upgrade...

Danct12, 2021-08-10:
Yes, please try to avoid that.

closed 2021-08-10 by Danct12