dreemurrs-embedded/Pine64-Arch #208: Bad checksums for 2021/07/13 release?

keith24, 2021-08-12:
I was able to confirm the checksums on the 20210628 release but not the 20210713.  I only tried pinephone phosh and pinephone barebone.  

Old ones work:
```sh
$ shasum -a 512 archlinux-pinephone-phosh-20210628.img.xz | grep 76de70e6e50e99df319a1656f8e33f739e04c4ae171184517485129bcc1826b0bcc5c034ddad4c4accc624b0f0416d180882edc85a45f328890af55264b58b08
76de70e6e50e99df319a1656f8e33f739e04c4ae171184517485129bcc1826b0bcc5c034ddad4c4accc624b0f0416d180882edc85a45f328890af55264b58b08  archlinux-pinephone-phosh-20210628.img.xz
$ shasum -a 512 archlinux-pinephone-barebone-20210628.img.xz | grep d965f2bfdddfd4d931ece0932a70312eb101406b8c4f8ae70d9f7a762ff74914f1aa0414256137f859d3a623ea920ea39765ea1496075c15f83c396e2881fd31
d965f2bfdddfd4d931ece0932a70312eb101406b8c4f8ae70d9f7a762ff74914f1aa0414256137f859d3a623ea920ea39765ea1496075c15f83c396e2881fd31  archlinux-pinephone-barebone-20210628.img.xz
```

New ones don't match:

```sh
$ shasum -a 512 archlinux-pinephone-phosh-20210713.img.xz
68ebeef415dee1df89bc7322bf9e3f7103708f39b425a33773a12613f56b6cc03598e12c8e1c344a40be32ab355a7022b06a709a58183119aaafdf82be17bfd3  archlinux-pinephone-phosh-20210713.img.xz
$ # should be 4c0ccc1ffe77bd0a223db05849ca6ee25db84f28ddd33cce2429e2822db5350498b5abc58b8c0bf5edd332a3c70fbc1ee65cb17a34d6a141a87a32f239024d1a
$ shasum -a 512 archlinux-pinephone-barebone-20210713.img.xz
81d6498dbd04237525d5b1de5c900f5c1bed6377aad714442ba59200a2827bf08fc046136ce55a14b27b5d80014706cf11af904de9bd283e8eaf3d9a2ad907e6  archlinux-pinephone-barebone-20210713.img.xz
$ # should be 7c539a5f9152631e9b75feb7b3bccf30dbb75efeda4b417782ff020be1c37d10ae7c7e63c51f0cb5f08df2f67970d558a767779480a8dba266aeb9c1abcdc031
```
I tried the download a few times... could someone confirm that the hashes are correct?

Danct12, 2021-08-13:
I've checked these two files, the hashes are correct for me.

```
$ wget https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20210713/archlinux-pinephone-phosh-20210713.img.xz https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20210713/archlinux-pinephone-barebone-20210713.img.xz
--2021-08-13 13:57:44--  https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20210713/archlinux-pinephone-phosh-20210713.img.xz
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving github.com (github.com)... 13.250.177.223
Connecting to github.com (github.com)|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-releases.githubusercontent.com/205039102/023ea380-e3d9-11eb-82e5-ece511dec5a4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210813%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210813T065744Z&X-Amz-Expires=300&X-Amz-Signature=362f4e72b6c6769100324dbb07cd96f85e08050d4410b224450b18ab9a2a1c82&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=205039102&response-content-disposition=attachment%3B%20filename%3Darchlinux-pinephone-phosh-20210713.img.xz&response-content-type=application%2Foctet-stream [following]
--2021-08-13 13:57:44--  https://github-releases.githubusercontent.com/205039102/023ea380-e3d9-11eb-82e5-ece511dec5a4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210813%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210813T065744Z&X-Amz-Expires=300&X-Amz-Signature=362f4e72b6c6769100324dbb07cd96f85e08050d4410b224450b18ab9a2a1c82&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=205039102&response-content-disposition=attachment%3B%20filename%3Darchlinux-pinephone-phosh-20210713.img.xz&response-content-type=application%2Foctet-stream
Resolving github-releases.githubusercontent.com (github-releases.githubusercontent.com)... 185.199.111.154, 185.199.108.154, 185.199.110.154, ...
Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.111.154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 695443136 (663M) [application/octet-stream]
Saving to: ‘archlinux-pinephone-phosh-20210713.img.xz’

archlinux-pinephone-phosh-2021 100%[=================================================>] 663.23M  19.8MB/s    in 33s     

2021-08-13 13:58:18 (19.9 MB/s) - ‘archlinux-pinephone-phosh-20210713.img.xz’ saved [695443136/695443136]

--2021-08-13 13:58:18--  https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20210713/archlinux-pinephone-barebone-20210713.img.xz
Connecting to github.com (github.com)|13.250.177.223|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-releases.githubusercontent.com/205039102/cf94ab00-e3d8-11eb-9fa2-83b3fa3633aa?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210813%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210813T065818Z&X-Amz-Expires=300&X-Amz-Signature=988fc1b8c4054446a03d83d140b283b349b66b8b73349038b38a31e031518f3f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=205039102&response-content-disposition=attachment%3B%20filename%3Darchlinux-pinephone-barebone-20210713.img.xz&response-content-type=application%2Foctet-stream [following]
--2021-08-13 13:58:18--  https://github-releases.githubusercontent.com/205039102/cf94ab00-e3d8-11eb-9fa2-83b3fa3633aa?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210813%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210813T065818Z&X-Amz-Expires=300&X-Amz-Signature=988fc1b8c4054446a03d83d140b283b349b66b8b73349038b38a31e031518f3f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=205039102&response-content-disposition=attachment%3B%20filename%3Darchlinux-pinephone-barebone-20210713.img.xz&response-content-type=application%2Foctet-stream
Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.111.154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 273715208 (261M) [application/octet-stream]
Saving to: ‘archlinux-pinephone-barebone-20210713.img.xz’

archlinux-pinephone-barebone-2 100%[=================================================>] 261.04M  18.9MB/s    in 14s     

2021-08-13 13:58:33 (18.8 MB/s) - ‘archlinux-pinephone-barebone-20210713.img.xz’ saved [273715208/273715208]

FINISHED --2021-08-13 13:58:33--
Total wall clock time: 49s
Downloaded: 2 files, 924M in 47s (19.6 MB/s)
$ shasum -a 512 archlinux-pinephone-phosh-20210713.img.xz archlinux-pinephone-barebone-20210713.img.xz
4c0ccc1ffe77bd0a223db05849ca6ee25db84f28ddd33cce2429e2822db5350498b5abc58b8c0bf5edd332a3c70fbc1ee65cb17a34d6a141a87a32f239024d1a  archlinux-pinephone-phosh-20210713.img.xz
7c539a5f9152631e9b75feb7b3bccf30dbb75efeda4b417782ff020be1c37d10ae7c7e63c51f0cb5f08df2f67970d558a767779480a8dba266aeb9c1abcdc031  archlinux-pinephone-barebone-20210713.img.xz
```

keith24, 2021-08-13:
`¯\_ (ツ)_/¯`

closed 2021-08-13 by keith24