dreemurrs-embedded/Pine64-Arch #210: Earphones do not work during phone calls until you insert them twice

kapinta, 2021-08-15:
Hello all.
After system reboot, when you make a phone call or receive it, if you insert earphones for the first time, you cannot hear/speak using them, but if you disconnect and connect the earphones again, from this moment they work fine again.
Thanks in advance for all your work.

d0n601, 2021-08-18:
I just thought they didn't work lol. Thanks for the top on connect/disconnect/reconnect.

tzcrawford, 2021-08-21:
I have not had the problem of having to insert earphones twice during a phone call. But I have experienced the other party not being able to hear me either from earphones, speaker phone, or regular phone mic. Usually at least one of the three works. In my case, when I make a phone call, only the left side produces sound while the right does not. After the phone call ends, both earphones work -- I am usually using Lollypop or something.