dreemurrs-embedded/Pine64-Arch #211: "No internet" on devices connected to pinephone hotspot

tzcrawford, 2021-08-22:
Using a command something like this: `sudo nmcli device wifi hotspot ifname wlan0 con-name Hotspot ssid something password something_else` should enable the hotspot feature. This has worked for me a couple weeks ago. But after some update, devices which connect to the hotspot say they have "no internet", meaning that they can connect to the pinephone's wifi but something is stopping the internet bridge between the modem connection and the hotspot. Also now, there is an option in the GNOME settings to make a hotspot, but it is grayed out. 

This could be due to a firewall or another security feature. The problem has occured before with pinephones using Manjaro: [https://forum.pine64.org/showthread.php?tid=10974&highlight=hotspot+internet](https://forum.pine64.org/showthread.php?tid=10974&highlight=hotspot+internet). Possibly this is due to the kernel being compiled with different options from before. I have tried downgrading the kernel from my cache before I first started experiencing this bug; but I could only revert to `linux-pine64 5.13.7-1` (~Aug 4th) from `linux-pine64 5.13.12-1`. But simply downgrading the linux kernel and headers did not solve the problem.

Danct12, 2021-09-15:
The issue seems to have partially fixed for me, internet access via hotspot works, but sometimes it doesn't. If it doesn't work I just need to toggle on/off.

Danct12, 2021-09-20:
I just took my PinePhone to a test drive today as a hotspot device and it works perfectly fine.

The first time enabling hotspot, there's no internet. But if you turn off hotspot and turn it back on it starts working.

closed 2021-09-20 by Danct12