dreemurrs-embedded/Pine64-Arch #214: chatty keeps crashing with omemo

dduenker, 2021-09-06:
Hi,

i just installed the os fresh, using the installer for encrypted fs: https://github.com/dreemurrs-embedded/archarm-mobile-fde-installer
Afterwards i installed `libpurple-lurch`, `libpurple-xmpp-http-upload` and `libpurple-carbons`.

Installed are: 
- chatty 0.3.4-1
- libpurple-lurch 0.6.9_dev-1
- libpurple-xmpp-http-upload 0.2.2-1
- libpurple-carbons 0.2.3-1

Chatty crashes, whenever sending or receiving omemo encrypted messages (when encryption is enabled).
I tried installing version 0.7.0-1 of `libpurple-lurch`, but it still crashed.

The message i get, when running chatty from the terminal is:
```
Segmentation fault (core dumped)
```

I can't make out where the core dump has been stored though.
Any idea, on how to fix this issue?

bogen85, 2021-09-06:
Interesting. I was wanting to use chatty with omemo, instead I've been using gajim. I'll try with the libpurple packages and see if it crashes for me as well.

bogen85, 2021-09-07:
I installed the aforementioned libpurple packages and restarted my phone. Upon restarting chatty I was not able to enable encryption.

dduenker, 2021-09-07:
@bogen85 
> I installed the aforementioned libpurple packages and restarted my phone. Upon restarting chatty I was not able to enable encryption.

I enabled it in a specific conversation via the button with the three dots in the top right corner. From there i could open the conversation settings, which had a slider to enable encryption.

Danct12, 2021-09-07:
Upstream issue: https://source.puri.sm/Librem5/chatty/-/issues/450

Danct12, 2021-09-07:
I've forked libolm (3548758ea96a09e1716f4b2b75692e20ce853cd2) with [this patch](https://gitlab.matrix.org/matrix-org/olm/-/commit/1b7973626e27beb05a55df6085f643a1323db987), can anyone test to see if it still crashes?

bogen85, 2021-09-07:
I'm still getting encryption not available.

Apart from these packages (I have omemo working in gajim) is there any other package I need?
* libpurple-lurch 0.6.9_dev-1
* libpurple-xmpp-http-upload 0.2.2-1
* libpurple-carbons 0.2.3-1

I would like to test this.

bogen85, 2021-09-08:
OK, I updated and got the new packages, restarted. Encryption now available, not able to trust keys like in gajim. Am able to enable encryption, but still not not able to decrypt incoming/sent messages. No crashes however.

bogen85, 2021-09-08:
"Recevied omemo message that does not contain a key for this device" is what I get when I try to send and encrypted message.

dduenker, 2021-09-08:
@Danct12 Encryption is working now on two devices. No crashes so far. Thanks a lot for the fast help. :)

@bogen85 The crashing has been solved, but you seem to be still having trouble with the encryption. Have you tried sending messages from as well as to chatty? I had a message not reach the other side, until i sent one in the other direction.

Danct12, 2021-09-29:
This is now fixed in libolm 3.2.5. We're now on 3.2.6!

closed 2021-09-29 by Danct12