dreemurrs-embedded/Pine64-Arch #217: Waydroid doesn't start on Barebone image

edp17, 2021-09-15:
I followed the instructions from here: https://fosstodon.org/@xaviers/106744248798164192
but got:
"modprobe: FATAL: Module binder_linux not found in directory /lib/modules/5.14.3-1-danctnix
modprobe: FATAL: Module ashmem_linux not found in directory /lib/modules/5.14.3-1-danctnix
Failed to stop anbox-container-manager.service: Unit anbox-container-manager.service not loaded.
/var/lib/lxc/anbox/anbox-net.sh: line 185: dnsmasq: command not found
Failed to setup anbox-net.
Failed to setup anbox-net.
umount: rootfs: not mounted.
chmod: cannot access '/run/user/1000/pulse': No such file or directory
chmod: cannot access '/dev/sw_sync': No such file or directory
chmod: cannot access '/sys/kernel/debug/sync/sw_sync': No such file or directory
chmod: cannot access '/dev/Vcodec': No such file or directory
chmod: cannot access '/dev/MTK_SMI': No such file or directory
chmod: cannot access '/dev/mdp_sync': No such file or directory
chmod: cannot access '/dev/mtk_cmdq': No such file or directory
chmod: cannot access '/dev/video32': No such file or directory
chmod: cannot access '/dev/video33': No such file or directory
chmod: cannot access '/dev/graphics/*': No such file or directory
lxc-start: anbox: cgroups/cgroup2_devices.c: bpf_program_load_kernel: 334 Function not implemented - Failed to load bpf program: (null)
lxc-start: anbox: network.c: netdev_configure_server_veth: 708 No such file or directory - Failed to attach "vethjfx4rp" to bridge "anbox0", bridge interface doesn't exist
lxc-start: anbox: network.c: lxc_create_network_priv: 3419 No such file or directory - Failed to create network device
lxc-start: anbox: start.c: lxc_spawn: 1826 Failed to create the network
lxc-start: anbox: start.c: __lxc_start: 2053 Failed to spawn container "anbox"
lxc-start: anbox: conf.c: run_buffer: 323 Script exited with status 126
lxc-start: anbox: start.c: lxc_end: 996 Failed to run lxc.hook.post-stop for container "anbox"
lxc-start: anbox: tools/lxc_start.c: main: 308 The container failed to start
lxc-start: anbox: tools/lxc_start.c: main: 313 Additional information can be obtained by setting the --logfile and --logpriority options"

Looks like the binder_linux module is missing for sure. Can you please add it to the image?
(Although there can be other missing stuff too.)

Danct12, 2021-09-15:
dnsmasq and nft seems to be missing, but can you also please try out those packages instead? https://danctnix.arikawa-hi.me/waydroid-test/

edp17, 2021-09-15:
Do you mean install all those? I will do it later because, in the meantime, I downloaded your phosh image to try waydroid.

Danct12, 2021-09-15:
Yes.

edp17, 2021-09-16:
I've gone back to the barebone image and installed all packages.
First, I got a 'waydroid-start: command not found' message, then I realised that in the last package you provided (waydroid-image-17.1_20210913-1-aarch64.pkg.tar.xz) there was only the system.img and vendor.img included. So, I installed the 'waydroid-20210730-1-aarch64.pkg.tar.xz' package first, then the other one 'waydroid-image-17.1_20210913-1-aarch64.pkg.tar.xz'.
Then I got this:

[root@danctnix alarm]# waydroid-start
modprobe: FATAL: Module binder_linux not found in directory /lib/modules/5.14.3-1-danctnix
modprobe: FATAL: Module ashmem_linux not found in directory /lib/modules/5.14.3-1-danctnix
Failed to stop anbox-container-manager.service: Unit anbox-container-manager.service not loaded.
/var/lib/lxc/anbox/anbox-net.sh: line 185: dnsmasq: command not found
Failed to setup anbox-net.
Failed to setup anbox-net.
umount: rootfs: not mounted.
chmod: cannot access '/run/user/1000/pulse': No such file or directory
chmod: cannot access '/dev/sw_sync': No such file or directory
chmod: cannot access '/sys/kernel/debug/sync/sw_sync': No such file or directory
chmod: cannot access '/dev/Vcodec': No such file or directory
chmod: cannot access '/dev/MTK_SMI': No such file or directory
chmod: cannot access '/dev/mdp_sync': No such file or directory
chmod: cannot access '/dev/mtk_cmdq': No such file or directory
chmod: cannot access '/dev/video32': No such file or directory
chmod: cannot access '/dev/video33': No such file or directory
chmod: cannot access '/dev/graphics/*': No such file or directory
lxc-start: anbox: cgroups/cgroup2_devices.c: bpf_program_load_kernel: 334 Function not implemented - Failed to load bpf program: (null)
lxc-start: anbox: network.c: netdev_configure_server_veth: 708 No such file or directory - Failed to attach "veth9rXNsS" to bridge "anbox0", bridge interface doesn't exist
lxc-start: anbox: network.c: lxc_create_network_priv: 3419 No such file or directory - Failed to create network device
lxc-start: anbox: start.c: lxc_spawn: 1826 Failed to create the network
lxc-start: anbox: start.c: __lxc_start: 2053 Failed to spawn container "anbox"
lxc-start: anbox: conf.c: run_buffer: 323 Script exited with status 126
lxc-start: anbox: start.c: lxc_end: 996 Failed to run lxc.hook.post-stop for container "anbox"
lxc-start: anbox: tools/lxc_start.c: main: 308 The container failed to start
lxc-start: anbox: tools/lxc_start.c: main: 313 Additional information can be obtained by setting the --logfile and --logpriority options

I am not sure that helps but I have not installed anything else on the barebone image. So, there is no GUI at all. Should I install a GUI? If, yes, which one do you recommend for the pinephone and waydroid?

Danct12, 2021-09-16:
Please do not install the old 20210730, there's a newer version in that repo (r30.15bbf62). You can install it by typing `sudo pacman -S waydroid`.

Waydroid **requires** a **Wayland** compositor, Phoc and Mutter seems to be the one that have been tested.

edp17, 2021-09-19:
I've installed Weston but didn't help either. I'll do a fresh start and try what you suggested. Thanks!

edp17, 2021-09-19:
I've installed all packages from https://danctnix.arikawa-hi.me/waydroid-test/ and waydroid and Phoc and then modified the boot.txt but then the command 'sudo waydroid-start' gave me "sudo: waydroid-start: command not found". I thought the shell scripts would be installed by 'sudo pacman -S waydroid'. What did I do wrong? :) Thanks.

Danct12, 2021-09-19:
Waydroid now uses a Python script to start, just type `waydroid --help` and you should be able to

edp17, 2021-09-20:
I have installed mutter, phoc and gdm. I've managed to get GUI where I could log in. There was the waydroid app icon but nothing happened, when I clicked on it. When I started it from the command line (ssh from PC) it didn't start either. (First I started 'sudo waydroid init -f', then 'sudo systemctl start waydroid-container' and finally I started the waydroid session-manager.)
Could you please let me know how you did install/configure the phoc, mutter and waydroid? Thanks.

Danct12, 2021-09-20:
Can you look into `sudo waydroid log`? As well as check `sudo waydroid logcat` when you launch the session.

Danct12, 2021-11-16:
Closing because lack of activities, but also can be user error. Feel free to open again if you're going to continue the discussion.

closed 2021-11-16 by Danct12