dreemurrs-embedded/Pine64-Arch #218: Mobile Internet broken after modemmanager and libmm-glib update

dariox86, 2021-09-15:
Since modemmanager-1.18.0-1 and libmm-glib-1.18.0-1 mobile Internet is completely unusable about 4 out of 5 times. Available coverage is almost invariably reported as 2G whereas it is supposed to be 4G. In fact no connection is available at all. Mobile signal strength seems correctly reported. Inbound and outbound calls seem to work as usual. Downgrading to modemmanager-1.16.8-1 and libmm-glib-1.16.8-1 fixes the issue. I am using  Biktorgj's open source modem firmware version 0.3.1.

Danct12, 2021-09-16:
4G mobile internet works fine on my end though, can you provide `journalctl -b`?

dariox86, 2021-09-16:
Attached the requested log with modemmanager-1.18.0-2 and libmm-glib-1.18.0-2. The moment I saved the log there was 4G network connectivity and it was working as expected, though the phone would report 2G instead. The reported signal quality seems accurate. Most of the time it is reported as 2G but there is no network connectivity at all, not even the reported 2G coverage.

[Log for journalctl -b](https://github.com/dreemurrs-embedded/Pine64-Arch/files/7181178/log.txt)

dariox86, 2021-09-18:
This is news from nine days ago: https://dylanvanassche.be/blog/2021/pinephone-modem-myths/ Please, read myth number 2. I don't know if this is still relevant though.

Danct12, 2021-09-19:
Don't think that's relevant, currently 1.18.0 is a stable release which can be safely upgraded to.

Anyway I looked at the log but can't find anything, maybe I missed something? Would be a smarter idea to have @DylanVanAssche looking at the log.

Danct12, 2021-09-19:
Hmm, after I looked again I found

```
set 16 23:38:58 danctnix ModemManager[3761]: <info>  [modem0/bearer2] couldn't start network: QMI protocol error (14): 'CallFailed'
set 16 23:38:58 danctnix ModemManager[3761]: <info>  [modem0/bearer2] verbose call end reason (2,210): [internal] pdn-ipv6-call-disallowed
```

After a look up, this seems to be relevant: https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/430

DylanVanAssche, 2021-09-27:
@Danct12 That might be indeed related...

dariox86, 2021-10-10:
It works as expected after reinstalling the operating system from scratch. Go figure...

dariox86, 2021-11-03:
I found a workaround and [opened a ticket](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/459) upstream.

dariox86, 2022-12-06:
This issue is gone for me in recent releases. The only problem I have though is that it takes a couple of minutes to connect to mobile network upon boot, but this is just a minor nuisance.

closed 2022-12-06 by dariox86