dreemurrs-embedded/Pine64-Arch #223: Check for lxc presence before initializing

raphj, 2021-09-23:
Hello,

I tried to run waydroid init while not having lxc installed. It issued an error and left waydroid in an inconsistent state where waydroid thought it was initialized but not really:

```
[18:29:14] Download https://sourceforge.net/projects/waydroid/files/images/system/lineage/waydroid_arm64/lineage-17.1-20210920-VANILLA-waydroid_arm64-system.zip/download
[18:34:08] Validating system image
[18:34:10] Extracting to /var/lib/waydroid/images
[18:36:41] Download https://sourceforge.net/projects/waydroid/files/images/vendor/waydroid_arm64/lineage-17.1-20210920-MAINLINE-waydroid_arm64-vendor.zip/download
[18:37:00] Validating vendor image
[18:37:00] Extracting to /var/lib/waydroid/images
[18:37:05] ERROR: LXC is not installed
[18:37:05] See also: <https://github.com/waydroid>
Run 'waydroid log' for details.

...
after installation of lxc
....
$ sudo waydroid container start
waydroid doesn't exist
[18:52:56] WayDroid container is
```

Maybe `wayland init` should be allowed to complete the initialization process after lxc is installed and/or check for lxc presence before downloading Android images?

Danct12, 2021-09-23:
How did you install Waydroid? The package on AUR has lxc as dependency.

raphj, 2021-09-23:
Whooops! I reported against the wrong repository. Sorry. I'm closing this issue.

(I installed waydroid on mobian using the official debian repository)

closed 2021-09-23 by raphj