dreemurrs-embedded/Pine64-Arch #235: Fragmented ecosystem

julianfairfax, 2021-11-04:
First off, I'd like to thank you for maintaining this OS as I believe Linux phones are the future despite their current issues, of which one of them I believe to be a fragmented ecosystem. There are many OSes for Linux phones with most of them having contributed something to most of the others. This is great from a development standpoint but not great from a user experience standpoint. The good thing is that most of the time this isn't really noticed, but in some cases it is quite obvious that different components from different developers are all being used in the same OS such as the green enter key in odk-sdl keyboard from postmarketOS or the green toggles in the tweaks app from postmarketOS. Both of these use the postmarketOS accent color, whose shade of green is only ever seen there in the OS, and is generally replaced by GNOME's blue accent color everywhere else. Of course this is a small issue in this case, and other cases exist such as Chatty from Purism, or King's Cross, Megapixels, and Portfolio from independent developers, but they do a better job of fitting in even though their developers aren't directly involved in this project either. I'm not suggesting that upstream developers don't tailor their components for their own OS, that each OS develop their own, or that one centralized entity take charge of all development, but rather that the distributor of the final OS, in this case you, tailor the components to match your final OS. For example, my suggestion here would be to use the GNOME accent color for the odk-sdl keyboard enter key and the tweaks app icon toggles. These might be small changes, even useless changes, but in my personal opinion at least, they're make the OS more appealing by eliminating what are small distractions from an otherwise seamless and functional OS. Again, thanks for making this, and any thoughts on this issue or changes for it are welcome.

julianfairfax, 2021-11-06:
A new logo would also be nice, something more simple than this one. There's just a bit too much going on in it for it be recognizable and easy on the eyes. For parity with the desktop it should also not show during boot, but this makes some sense on a mobile phone.

julianfairfax, 2021-11-06:
If anyone wants to do all these things manually you can run these commands:
```
sudo sed -i 's/003C00/3584E4/' /etc/osk.conf 
sudo mkinitcpio -P
sudo curl -L -s -o /usr/share/icons/hicolor/scalable/apps/org.danctnix.Tweaks.svg https://gitlab.gnome.org/GNOME/gnome-tweaks/-/raw/master/data/org.gnome.tweaks.svg
sudo sed -i 's/quiet\ bootsplash\.bootfile\=bootsplash\-themes\/danctnix\/bootsplash//' /boot/boot.txt
cd /boot
sudo ./mkscr 
```
You'll have to rerun them when something is upgraded, and of course you could just build the packages yourself instead of doing this or changing the upstream, but it would be a welcome change in my opinion if things were more polished by default.

julianfairfax, 2021-11-06:
Posting this here as an idea for a better icon. Inspired by the icon of the Mobian Matrix room which I found to have beauty in it's simplicity. I can send over the project file if you want but it's not difficult to replicate
[image]
.

1peter10, 2021-11-06:
I read the initial post (please consider using paragraphs for readability) yesterday, and took some time to think about it:

- regarding postmarketOS/DanctNIX Tweaks: I don't find the green that disturbing, and since GNOME Tweaks has also been gotten some libhandy "treatment" (sadly it's still to wide), I am against changing the icon to the one of GNOME Tweaks - otherwise people that have both installed would end up with to identical icons named Tweaks, that are different apps and have different features.
- OSK-SDL: Really? How often do you reboot your phone? [Purism have themed OSK-SDL](https://source.puri.sm/librem5-apps/osk-sdl/-/merge_requests/9) to match Squeekboard better, so the hard work is done.

I recommend trying to change the relevant PKGBUILD scripts and creating pull requests so that Danct12 has an easier time if he wants to apply these changes.

julianfairfax, 2021-11-06:
> I read the initial post (please consider using paragraphs for readability) yesterday, and took some time to think about it:
> 
There are now paragraphs
> * regarding postmarketOS/DanctNIX Tweaks: I don't find the green that disturbing, and since GNOME Tweaks has also been gotten some libhandy "treatment" (sadly it's still to wide), I am against [...]

Well they could also have all three installed, but realistically it's going to be the one best supported: danctnix's
>     * OSK-SDL: Really? How often do you reboot your phone? [Purism have themed OSK-SDL](https://source.puri.sm/librem5-apps/osk-sdl/-/merge_requests/9) to match Squeekboard better, so the hard work is done.
> 
> 

Quite often actually, this is a development device after all. And the hard work is changing one config file.
> I recommend trying to change the relevant PKGBUILD scripts and creating pull requests so that Danct12 has an easier time if he wants to apply these changes.

I already made a pull request for danctnix tweaks and will make one here for osk-sdl if I get a positive response from them here.

Danct12, 2021-11-06:
Regarding DanctNIX Tweaks, I'll get rid of it when pmOS tweaks becomes distro-agnostic - upstream issue: https://gitlab.com/postmarketOS/postmarketos-tweaks/-/issues/12

julianfairfax, 2021-11-06:
I suppose that goes for osk-sdl too with this upstream issue I opened: https://gitlab.com/postmarketOS/osk-sdl/-/issues/137

julianfairfax, 2021-11-07:
Regarding logos, here's an updated proposition for the Matrix room, still inspired by Mobian:
[image]

This wouldn't work during boot so I have a proposition for that, inspired by Arch Linux ARM:
[image]

In case you noticed, that one wouldn't work on boot either, so the actual proposition has a different colour for the text. I've attached it here:
https://user-images.githubusercontent.com/43753131/140637055-cfe61aa0-dd5a-4529-ae86-a472c82c0221.png

julianfairfax, 2021-11-07:
@Danct12 Should I open a separate issue or a pull request for the branding changes or do you not any changes to be made? Cause this issue is closed apart from that, which could be considered a separate issue and which you haven't yet commented on. I suggest you consider branding changes as the logo is literally the first thing you see when you turn on the phone and the last thing you see what it turns off, and I believe that Mobian has excelled at this branding, hence my inspiration for the Matrix room logo. Their boot screen logo is also similar to my second proposition, in the way that it's just text, but this one is obviously from the Arch Linux ARM logo. I believe a good brand is important as it makes the experience feel more professional. If you have conflicting views, please close this issue.

mossymossmoss, 2022-02-15:
I created an issue to talk about a logo change (#314) without seeing that this issue existed. Here are some mock-ups of possible logos that I made with variants that include a phone camera bar

[image]
[image]
[image]
[image]

julianfairfax, 2022-02-15:
> I created an issue to talk about a logo change (#314) without seeing that this issue existed. Here are some mock-ups of possible logos that I made with variants that include a phone camera bar
> 
> [image]
> [image]
> [image]
> [image]

Those actually really nice!

closed 2022-08-09 by julianfairfax