dreemurrs-embedded/Pine64-Arch #243: sxmo locale issue

K3nn3th2, 2021-11-17:
- Device: PineTab
- Kernel Version : 5.14.13-1-danctnix
- UI: sxmo

### Steps to reproduce
- sudo pacman -S mcomix
- mcomix

### Expected behavior
- mcomix should start

### Actual behavior
Python3.9 throws an error:
Error: unsupported locale setting

### Logfiles and additional information

ArenM, 2021-11-22:
I ran into this today, and it appears there is no locale configuration included with the current sxmo image. The solution this is to either [configure a locale](https://wiki.archlinux.org/title/Locale), or install the `glibc-locales` package.

3nprob, 2021-12-03:
How about the following?

1. First bootup has script that picks up and configures locale from static file (so image can be easily prepared before first boot)
2. Set a default locale (e.g. `en_US.UTF-8` or `en_DK.UTF-8`).  User can always change.
3. Prompt user for locale on first boot (but before X/Wayland

IMO static file, with preconfigured default, makes most sense

This should be useful for all, not just sxmo users

Danct12, 2022-02-13:
glibc-locales has been added into the sxmo image by default for a few releases now. You can install the package or manually configure the locale.

closed 2022-02-13 by Danct12