dreemurrs-embedded/Pine64-Arch #244: SMS unstable on Danish sim-card

johays, 2021-11-17:
- Device: PinePhone
- Kernel Version : 5.15.2-1-danctnix
- UI: Phosh 0.14.0-1

Chatty (v 0.4.0-1) unable to send sms on Danish number. A week ago it was working fine. For a week it has been impossible to get messages through. SIM-card has been tested on another phone and is working fine.

### Steps to reproduce
Reproduce by sending SMS from a Danish sim-card. 

Both without contry-cocde and with +45 and 0045 contry codes have been tried, with no positive result.

### Expected behavior
That messages are send, received by the recipient and that a delivery-reports gets back to Chatty.

### Actual behavior
Messages are not send and an error is represented by a small red cross. 

### Logfiles and additional information
(please direct me to relevant logfiles and/or debug-info, and I will happily provide it)

johays, 2021-11-18:
(Relating to [this bug in chatty](https://source.puri.sm/Librem5/chatty/-/issues/505), I have tried forcing 2G and 3G, but it doesn't seem to resolve the issue)

Danct12, 2021-11-30:
It seems like both G and alexela from the chat has also reported the same issue. It looks like removing the battery, wait for a bit and put it back in fixed the issue.. very weird one. Can you give it a shot?

Danct12, 2023-09-07:
Closing as this issue has been inactive for ~2 years, as well as lack of response from the reporter. If this is still an issue, please open a new bug report.

closed 2023-09-07 by Danct12