dreemurrs-embedded/Pine64-Arch #254: PPP can't start EG25

fbmoose48, 2021-12-10:
eg25-manager can't start modem. seems like maybe bad config on Arch? works on Mobian, so not hardware.

- Device: PinePhone Pro
- Kernel Version : 5.16
- UI: Phosh

### Steps to reproduce
eg25-manager can't start modem

### Expected behavior
cellular service

### Actual behavior
no cellular service

### Logfiles and additional information

`mmcli -m any --command='AT+QNWINFO'`

returns `error: couldn't find modem`
`journalctl -u eg25-manger`

`Nov 29 00:00:25 danctnix eg25manager[4005]: unable to find a suitable config file!
Nov 29 00:00:28 danctnix systemd[1]: eg25-manager.service: Main process exited, code=dumped, status=5/TRAP
Nov 29 00:00:28 danctnix systemd[1]: eg25-manager.service: Failed with result 'core-dump'.
Nov 29 00:00:28 danctnix systemd[1]: eg25-manager.service: Scheduled restart job, restart counter is at 4.
Nov 29 00:00:28 danctnix systemd[1]: Stopped Quectel EG25 modem.
Nov 29 00:00:28 danctnix systemd[1]: Started Quectel EG25 modem. `

Danct12, 2021-12-10:
I already got it working and it should work soon with a new kernel.

bjadel, 2021-12-10:
Same problem here an my PPP. 

Kernel: 
```
[alarm@danctnix ~]$ uname -a
Linux danctnix 5.16.0-rc2-2-danctnix #1 SMP PREEMPT Thu Dec 9 12:05:21 UTC 2021 aarch64 GNU/Linux`
```

I have enabled the systemd unit `eg25-manager` and after that I get this in the journal:
```
Dez 10 14:06:38 danctnix systemd[1]: Started Quectel EG25 modem.
Dez 10 14:06:39 danctnix eg25manager[3072]: Opening config file: /usr/share/eg25-manager/pine64,pinephone-pro.toml
Dez 10 14:06:39 danctnix eg25manager[3072]: Unable to configure /dev/ttyS3
Dez 10 14:06:39 danctnix eg25manager[3072]: Starting modem...
Dez 10 14:06:40 danctnix eg25manager[3072]: Executed power-on/off sequence
Dez 10 14:06:40 danctnix eg25manager[3072]: taking systemd sleep inhibitor
Dez 10 14:06:40 danctnix eg25manager[3072]: oFono vanished from D-Bus
Dez 10 14:06:40 danctnix eg25manager[3072]: inhibitor sleep fd is 20
Dez 10 14:06:41 danctnix eg25manager[3072]: ModemManager appeared on D-Bus
Dez 10 14:07:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:07:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:08:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:08:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:09:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:09:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:10:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:10:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:11:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:11:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:12:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:12:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:13:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:13:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:14:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:14:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:15:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:15:38 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:16:08 danctnix eg25manager[3072]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:16:15 danctnix eg25manager[3072]: Request to quit...
Dez 10 14:16:15 danctnix eg25manager[3072]: g_source_remove: assertion 'tag > 0' failed
Dez 10 14:16:15 danctnix eg25manager[3072]: g_array_free: assertion 'array' failed
Dez 10 14:16:15 danctnix eg25manager[3072]: g_array_free: assertion 'array' failed
Dez 10 14:16:15 danctnix eg25manager[3072]: g_array_free: assertion 'array' failed
Dez 10 14:16:15 danctnix eg25manager[3072]: g_array_free: assertion 'array' failed
Dez 10 14:16:15 danctnix systemd[1]: Stopping Quectel EG25 modem...
Dez 10 14:16:15 danctnix eg25manager[3072]: dropping systemd sleep delay inhibitor
Dez 10 14:16:15 danctnix eg25manager[3072]: Modem down, quitting...
Dez 10 14:16:15 danctnix systemd[1]: eg25-manager.service: Deactivated successfully.
Dez 10 14:16:15 danctnix systemd[1]: Stopped Quectel EG25 modem.
Dez 10 14:16:15 danctnix systemd[1]: Started Quectel EG25 modem.
Dez 10 14:16:15 danctnix eg25manager[4132]: Opening config file: /usr/share/eg25-manager/pine64,pinephone-pro.toml
Dez 10 14:16:15 danctnix eg25manager[4132]: Unable to configure /dev/ttyS3
Dez 10 14:16:15 danctnix eg25manager[4132]: Starting modem...
Dez 10 14:16:16 danctnix eg25manager[4132]: Executed power-on/off sequence
Dez 10 14:16:16 danctnix eg25manager[4132]: taking systemd sleep inhibitor
Dez 10 14:16:16 danctnix eg25manager[4132]: ModemManager appeared on D-Bus
Dez 10 14:16:16 danctnix eg25manager[4132]: oFono vanished from D-Bus
Dez 10 14:16:16 danctnix eg25manager[4132]: inhibitor sleep fd is 21
Dez 10 14:16:45 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:17:15 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:17:45 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:18:15 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:18:45 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:19:15 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
Dez 10 14:19:45 danctnix eg25manager[4132]: Rescheduling upload since modem isn't online yet, in 30s
```

fbmoose48, 2021-12-10:
my kernel is currently 5.16.0-rc2-2-danctnix

Danct12, 2021-12-10:
This is now fixed with `5.16.0-rc3-1`

closed 2021-12-10 by Danct12

fbmoose48, 2021-12-10:
agreed. thanks!

fedov, 2021-12-10:
Unfortunately this is still happening on my PP (not pro) with kernel  5.16.0-rc3-1. I got it working again by downgrading eg25-manager to 0.4.1-1. With eg25-manager-0.4.2-1 I experience the same behavior as described above...

Danct12, 2021-12-11:
Still works for me on my PinePhone 1.2a. If you can, please open a bug report on https://gitlab.com/mobian1/devices/eg25-manager/-/issues

fedov, 2021-12-11:
This should be the PinePhone 1.2b. By the way - Is there a way to detect the hardware revision programmatically?

Danct12, 2021-12-11:
It should be autodetected automatically via U-Boot. You can check if it's using v1.2 dtb by running `cat /proc/device-tree/model`.

But I highly recommend opening a bug report on upstream eg25-manager and give them as much information as you can if this issue fixed itself in the previous version.

bjadel, 2021-12-11:
Now on Pinephone Pro works very well. I have receive a sms and the mobile internet works, too. Thank you.

fbmoose48, 2021-12-11:
try making a call or playing audio. If you have a working ucm config please share, I have opened #255