dreemurrs-embedded/Pine64-Arch #255: No audio/microphone capabilities

fbmoose48, 2021-12-10:
- Device: PinePhone Pro
- Kernel Version :  5.16.0-rc3-1
- UI: phosh

### Steps to reproduce
place/answer a call, play video in firefox

### Expected behavior
audio playback, microphone input

### Actual behavior
can'y hear or be heard on calls (both regular and speaker), no audio playback in browser.

"Sound" in settings indicates "Speaker" as output device. testing speakers gets no output. Playing audio in browswer gets nooutput, although the bar underneath does move as though it is detecting audio to output.

"Sound" in settings indicates "Analog Input" as input device. Not able to be heard on calls. "Sound Recorder" records a flat line with no sound when played back.

### Logfiles and additional information
`alarm@danctnix ~]$ journalctl | grep callaudiod`
`Dec 10 14:40:13 danctnix audit[3409]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=3409 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:40:13 danctnix kernel: audit: type=1701 audit(1639165213.733:826): auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=3409 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:40:38 danctnix systemd-coredump[142893]: Process 3409 (callaudiod) of user 1000 dumped core.`
`                                                   Found module callaudiod with build-id: c0a532b2251beabf5437c88cd8214e5e36bff9dd`
`                                                   #1  0x0000aaaac6ef6ba0 n/a (callaudiod + 0x6ba0)`
`                                                   #14 0x0000aaaac6ef30d4 main (callaudiod + 0x30d4)`
`                                                   #16 0x0000aaaac6ef3130 n/a (callaudiod + 0x3130)`
`                                                   #17 0x0000aaaac6ef3130 n/a (callaudiod + 0x3130)`
`Dec 10 14:48:01 danctnix audit[142908]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=142908 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:48:01 danctnix kernel: audit: type=1701 audit(1639165681.503:849): auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=142908 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:48:03 danctnix systemd-coredump[143019]: Process 142908 (callaudiod) of user 1000 dumped core.`
`                                                   Found module callaudiod with build-id: c0a532b2251beabf5437c88cd8214e5e36bff9dd`
`                                                   #1  0x0000aaaad0836ba0 n/a (callaudiod + 0x6ba0)`
`                                                   #14 0x0000aaaad08330d4 main (callaudiod + 0x30d4)`
`                                                   #16 0x0000aaaad0833130 n/a (callaudiod + 0x3130)`
`                                                   #17 0x0000aaaad0833130 n/a (callaudiod + 0x3130)`

`alarm@danctnix ~]$ journalctl | grep callaudiod`
`Dec 10 14:40:13 danctnix audit[3409]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=3409 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:40:13 danctnix kernel: audit: type=1701 audit(1639165213.733:826): auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=3409 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:40:38 danctnix systemd-coredump[142893]: Process 3409 (callaudiod) of user 1000 dumped core.`
`                                                   Found module callaudiod with build-id: c0a532b2251beabf5437c88cd8214e5e36bff9dd`
`                                                   #1  0x0000aaaac6ef6ba0 n/a (callaudiod + 0x6ba0)`
`                                                   #14 0x0000aaaac6ef30d4 main (callaudiod + 0x30d4)`
`                                                   #16 0x0000aaaac6ef3130 n/a (callaudiod + 0x3130)`
`                                                   #17 0x0000aaaac6ef3130 n/a (callaudiod + 0x3130)`
`Dec 10 14:48:01 danctnix audit[142908]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=142908 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:48:01 danctnix kernel: audit: type=1701 audit(1639165681.503:849): auid=1000 uid=1000 gid=1000 ses=2 subj==unconfined pid=142908 comm="callaudiod" exe="/usr/bin/callaudiod" sig=11 res=1`
`Dec 10 14:48:03 danctnix systemd-coredump[143019]: Process 142908 (callaudiod) of user 1000 dumped core.`
`                                                  Found module callaudiod with build-id: c0a532b2251beabf5437c88cd8214e5e36bff9dd`
`                                                   #1  0x0000aaaad0836ba0 n/a (callaudiod + 0x6ba0)`
`                                                   #14 0x0000aaaad08330d4 main (callaudiod + 0x30d4)`
`                                                   #16 0x0000aaaad0833130 n/a (callaudiod + 0x3130)`
`                                                   #17 0x0000aaaad0833130 n/a (callaudiod + 0x3130)`

bjadel, 2021-12-11:
Same here. Do you need more information?

fbmoose48, 2021-12-11:
I'm happy to post more logs if that helps, just need some direction in what you'd even need.

I understand Mobian has a working config at
https://gitlab.com/mobian1/packages/alsa-ucm-conf/-/blob/mobian/debian/patches/0005-ucm2-add-PinePhone-Pro-configuration.patch

Danct12, 2021-12-11:
Audio is heavily a WIP now, that UCM works, but you will not be able to get any callaudio from it now.

Danct12, 2021-12-11:
`alsa-ucm-pinephonepro` has been added.

closed 2021-12-11 by Danct12

fbmoose48, 2021-12-11:
Soundrecorder and audio playback works, do calls work as well? I can test calls out with a SIM  switch if that's unknown

bjadel, 2021-12-11:
> Soundrecorder and audio playback works, do calls work as well? I can test calls out with a SIM switch if that's unknown

Soundrecorder and sound tests in the settings audio view works very well. But calls doesn't work by me.