dreemurrs-embedded/Pine64-Arch #256: Megapixel doesn't work

bjadel, 2021-12-11:
- Device: PinePhone Pro
- Kernel: 5.16.0-rc3-1-danctnix
- UI: phosh

```
/usr/share/megapixels/config/pine64,pinephone-pro.ini not found
Could not find any config file
```

Danct12, 2021-12-11:
Please report to upstream: https://gitlab.com/postmarketOS/megapixels/-/issues

bjadel, 2021-12-12:
Issue: https://gitlab.com/postmarketOS/megapixels/-/issues/33

ewtoombs, 2024-03-06:
I just noticed errors in the dmesg:

```
[    6.554957] mc: Linux media interface: v0.10
[    6.621764] videodev: Linux video capture interface: v2.00
[    6.715142] NET: Registered PF_BLUETOOTH protocol family
[    6.736700] ov8858 1-0036: supply avdd not found, using dummy regulator
[    6.737237] ov8858 1-0036: supply dvdd not found, using dummy regulator
[    6.737446] ov8858 1-0036: xvclk mismatched, modes are based on 24MHz
[    6.739888] imx258 1-001a: supply vana not found, using dummy regulator
[    6.740237] imx258 1-001a: supply vdig not found, using dummy regulator
[    6.741702] ov8858 1-0036: Detected OV8858 sensor, revision 0xb2
[    6.797747] rockchip_vdec: module is from the staging directory, the quality is unknown, you have been warned.
[    6.803773] rockchip-rga ff680000.rga: HW Version: 0x03.02
[    6.845478] rockchip-rga ff680000.rga: Registered rockchip-rga as /dev/video0
[    6.849691] rkvdec ff660000.video-codec: Adding to iommu group 2
[    6.852451] hantro-vpu ff650000.video-codec: Adding to iommu group 3
[    6.897965] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    6.900345] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    6.900581] hantro-vpu ff650000.video-codec: registered rockchip,rk3399-vpu-enc as /dev/video2
[    6.901365] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[    6.901746] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    6.901777] cfg80211: failed to load regulatory.db
[    6.908827] hantro-vpu ff650000.video-codec: registered rockchip,rk3399-vpu-dec as /dev/video3
[    6.911166] dw_wdt ff848000.watchdog: No valid TOPs array specified
[    7.016919] dw-apb-uart ff180000.serial: failed to request DMA
[    7.043616] input: gpio-vibrator as /devices/platform/vibrator/input/input5
[    7.164806] simple-amplifier audio-amplifier: supply VCC not found, using dummy regulator
[    7.242439] rkisp1 ff910000.isp0: Adding to iommu group 4
[    7.242829] genirq: Flags mismatch irq 42. 00000004 (rkisp1) vs. 00000084 (ff914000.iommu)
[    7.251108] CPU: 3 PID: 3369 Comm: (udev-worker) Tainted: G         C         6.7.4-2-danctnix #1 58478a8a31aa404a66ad1aed92eb6314e0a70955
[    7.251133] Hardware name: Pine64 PinePhonePro (DT)
[    7.251138] Call trace:
[    7.251143]  dump_backtrace+0xb0/0x120
[    7.251167]  show_stack+0x20/0x38
[    7.251178]  dump_stack_lvl+0x48/0x60
[    7.260441]  dump_stack+0x18/0x28
[    7.260458]  __setup_irq+0x418/0x6c0
[    7.260473]  request_threaded_irq+0xf4/0x1b8
[    7.260485]  devm_request_threaded_irq+0x88/0x110
[    7.260499]  rkisp1_probe+0x14c/0x7f8 [rockchip_isp1 85a0eca46d9e830ec0faa8978b75fceb1f8d96c3]
[    7.260556]  platform_probe+0x70/0xd0
[    7.260571]  really_probe+0x18c/0x3d8
[    7.260586]  __driver_probe_device+0x84/0x180
[    7.260600]  driver_probe_device+0x48/0xf0
[    7.260614]  __driver_attach+0xfc/0x210
[    7.260627]  bus_for_each_dev+0x80/0xe8
[    7.260640]  driver_attach+0x2c/0x40
[    7.260653]  bus_add_driver+0x118/0x228
[    7.260667]  driver_register+0x68/0x138
[    7.260677]  __platform_driver_register+0x30/0x48
[    7.260688]  rkisp1_drv_init+0x28/0xff8 [rockchip_isp1 85a0eca46d9e830ec0faa8978b75fceb1f8d96c3]
[    7.260732]  do_one_initcall+0x60/0x300
[    7.260743]  do_init_module+0x60/0x218
[    7.260756]  load_module+0x215c/0x2238
[    7.260767]  init_module_from_file+0x90/0xf0
[    7.260779]  __arm64_sys_finit_module+0x20c/0x320
[    7.260791]  invoke_syscall+0x78/0x108
[    7.260804]  el0_svc_common.constprop.0+0xc8/0xf0
[    7.260816]  do_el0_svc+0x24/0x38
[    7.260828]  el0_svc+0x28/0xe0
[    7.260842]  el0t_64_sync_handler+0x100/0x130
[    7.260852]  el0t_64_sync+0x158/0x160
[    7.264392] rkisp1 ff910000.isp0: request irq failed: -16
[    7.274890] rkisp1: probe of ff910000.isp0 failed with error -16
[    7.275891] rkisp1 ff920000.isp1: Adding to iommu group 5
[    7.283714] genirq: Flags mismatch irq 43. 00000004 (rkisp1) vs. 00000084 (ff924000.iommu)
[    7.292162] CPU: 1 PID: 3369 Comm: (udev-worker) Tainted: G         C         6.7.4-2-danctnix #1 58478a8a31aa404a66ad1aed92eb6314e0a70955
[    7.292190] Hardware name: Pine64 PinePhonePro (DT)
[    7.292196] Call trace:
[... 17 lines cut ...]
[    7.301722]  __platform_driver_register+0x30/0x48
[    7.301732]  rkisp1_drv_init+0x28/0xff8 [rockchip_isp1 85a0eca46d9e830ec0faa8978b75fceb1f8d96c3]
[    7.301775]  do_one_initcall+0x60/0x300
[    7.301786]  do_init_module+0x60/0x218
[    7.301799]  load_module+0x215c/0x2238
[    7.301811]  init_module_from_file+0x90/0xf0
[    7.301822]  __arm64_sys_finit_module+0x20c/0x320
[    7.301834]  invoke_syscall+0x78/0x108
[    7.301848]  el0_svc_common.constprop.0+0xc8/0xf0
[    7.301860]  do_el0_svc+0x24/0x38
[    7.301871]  el0_svc+0x28/0xe0
[    7.301885]  el0t_64_sync_handler+0x100/0x130
[    7.301895]  el0t_64_sync+0x158/0x160
[    7.303472] rkisp1 ff920000.isp1: request irq failed: -16
[    7.313999] rkisp1: probe of ff920000.isp1 failed with error -16
```

rkisp was unable to request its irq.
Nothing else is going to work without that.
I've found the same problem here:

https://lore.kernel.org/all/87o7eo8vym.fsf@gmail.com/T/

But that thread says the issue was fixed.
I'm running linux-megi 6.7.4-2.

ewtoombs, 2024-03-07:
The issue *was* fixed, but it didn't make it into the `orange-pi-6.7-20240205-2255` tag, which is the one being used by linux-megi 6.7.4-2.
I patched it myself, and that *particular* problem is now fixed.
Now, I'm trying to get literally any output out of the thing.
v4l2-ctl --stream-mmap keeps giving me "VIDIOC_STREAMON returned -1 (Broken pipe)".

ewtoombs, 2024-03-11:
I fixed the broken pipe thing.

This is the script I am using right now:
```
#!/bin/bash -xe

# These can normally be retrieved with the bus info string, but the imx258 and ov8858 have the same bus info strings!
media_dev='/dev/media3'
imx258_rkisp1_mainpath='/dev/video8'

x=4208
y=3120

# These three pixel formats have to match.
# Can I just say that it is a little ridiculous that these don't all follow the same naming convention.
mbus_pixelformat='SRGGB10_1X10'
v4l2_pixelformat='RG10'
ffmpeg_pixelformat='bayer_rggb16be'

mbus_fmt="fmt:$mbus_pixelformat/${x}x$y"
crop="crop:(0,0)/${x}x$y"

media-ctl --verbose -d "$media_dev" --reset

media-ctl --verbose -d "$media_dev" --set-v4l2 "'imx258 1-001a':0 [$mbus_fmt]"
media-ctl --verbose -d "$media_dev" --set-v4l2 "'rkisp1_csi':0 [$mbus_fmt]"
media-ctl --verbose -d "$media_dev" --set-v4l2 "'rkisp1_csi':1 [$mbus_fmt]"
media-ctl --verbose -d "$media_dev" --set-v4l2 "'rkisp1_isp':0 [$mbus_fmt $crop]"
media-ctl --verbose -d "$media_dev" --set-v4l2 "'rkisp1_isp':2 [$mbus_fmt $crop]"
media-ctl --verbose -d "$media_dev" --set-v4l2 "'rkisp1_resizer_mainpath':0 [$mbus_fmt $crop]"
media-ctl --verbose -d "$media_dev" --set-v4l2 "'rkisp1_resizer_mainpath':1 [$mbus_fmt]"
v4l2-ctl --verbose -d "$imx258_rkisp1_mainpath" --set-fmt-video "width=$x,height=$y,pixelformat=$v4l2_pixelformat"

media-ctl --verbose -d "$media_dev" -l "'imx258 1-001a':0 -> 'rkisp1_csi':0 [1]"
media-ctl --verbose -d "$media_dev" -l "'rkisp1_csi':1 -> 'rkisp1_isp':0 [1]"
media-ctl --verbose -d "$media_dev" -l "'rkisp1_isp':2 -> 'rkisp1_resizer_mainpath':0 [1]"

#sleep 10
v4l2-ctl --verbose -d "$imx258_rkisp1_mainpath" --stream-mmap --stream-count 1 --stream-to=frame.raw

ffmpeg -y -s:v "${x}x$y" -pix_fmt "$ffmpeg_pixelformat" -i frame.raw frame.png
```
Now, I'm getting a hexagonal array of isolated black pixels.
Also, the bit depth is advertised at 10 bits per pixel, but the camera only outputs 8 bits per pixel.

Here is an example hexdump:
```

00000000: 1500 1800 1200 1700 1200 1800 1400 1600  ................
00000010: 1200 1500 1200 1800 1400 1700 1300 1700  ................
00000020: 1400 1700 1500 1800 1400 1a00 1400 1600  ................
00000030: 1300 1900 1400 1600 1500 1700 1300 1900  ................
00000040: 1300 1900 1300 1900 1300 1600 1200 1600  ................
00000050: 1500 1500 1300 1600 1400 1700 1500 1800  ................
00000060: 1500 1500 1400 1600 1500 1500 1200 1800  ................
00000070: 1500 1700 1500 1800 1300 1800 1600 1800  ................
00000080: 1400 1800 1300 1900 1400 1800 1300 1700  ................
00000090: 1300 1600 1500 1600 1400 1700 1300 1a00  ................
```
Every other byte is all zeros.
There is supposed to be two bits of data where these zero bytes are.

There is like 6 different bugs here.
I need to split them off into separate issues.
Many of them don't even have anything to do with Pine64-Arch.