dreemurrs-embedded/Pine64-Arch #263: Add support for eMMC speed mod as in postmarketOS

dariox86, 2021-12-25:
Feature request: implement [eMMC speed hack](https://izzo.pro/pinephone-vccq-mod/) as in [postmarketOS](https://gitlab.com/postmarketOS/pmaports/-/merge_requests/2665).

Danct12, 2022-01-13:
U-Boot is updated to implement user script (which will be needed for this mod), please diff your boot.txt with the pacnew version.

dariox86, 2022-01-13:
Thank you. I will apply the hardware modification and report here. By the way there is no .pacnew file. Maybe it was just overwritten because the original was unmodified.

Danct12, 2022-01-14:
It it on 2022.01? Anyway, do the mod at your own risk.

dariox86, 2022-01-14:
> It it on 2022.01? Anyway, do the mod at your own risk.

Sorry, I don't understand. The system is up to date. I got the updated U-Boot. It should work, right? At my own risk, of course.

dariox86, 2022-01-14:
It's not working. It doesn't boot past the Tux screen with the spinning wheel.

Danct12, 2022-01-16:
You need to drop the attached file to /boot partition. Can you also send me your boot.scr as well?

[vccq.tar.gz](https://github.com/dreemurrs-embedded/Pine64-Arch/files/7877640/vccq.tar.gz)

Danct12, 2022-01-16:
If you're on 2022.01, the LED should be green on boot up.

You also need to push these files (attached below) to the /boot partition for it to work. Just in case, can you attach your boot.scr in the comment?

[vccq.tar.gz](https://github.com/dreemurrs-embedded/Pine64-Arch/files/7877652/vccq.tar.gz)

dariox86, 2022-01-16:
My LED is green on boot. I had already tried copying these two files over from postmartketOS to no avail. Either way both with your files and with the files I had tried I end up in a boot loop:

[image]

Attached my boot.scr:
[boot.scr.zip](https://github.com/dreemurrs-embedded/Pine64-Arch/files/7877732/boot.scr.zip)

I also have issues with JumpDrive 0.8: after reading or writing a few megabytes my PinePhone units disappear from the operating system due to an unspecified I/O issue.

Danct12, 2022-01-16:
Hmm, seems like I made a mistake when setting the location where the user script should be in memory.

Can you modify boot.txt and change `user_scriptaddr` memory region to `0x61dbc200` then regenerate boot.scr and see if it works?

dariox86, 2022-01-16:
It works. Thank you!

closed 2022-01-16 by dariox86

sorry-i-am-late, 2023-11-09:
> Hmm, seems like I made a mistake when setting the location where the user script should be in memory.
> 
> Can you modify boot.txt and change `user_scriptaddr` memory region to `0x61dbc200` then regenerate boot.scr and see if it works?

how do I regenerate my boot.scr? 
I performed this mod on my ubports 2gb/16gb (Kimtigo chip) and nothing changed in terms of write speed. (nothing is worse than before)
Does the need to replace `0x50100000` with `0x61dbc200` not apply to the older emmc?

dariox86, 2023-11-11:
My PinePhone had lost both mobile and Wi-Fi connection a while ago and I became uninterested ever since. Actually I regret having bought one. I was prepared to bear with software problems, but I did not expect such hardware issues. Anyway, I remember I just needed to add two files to /boot partition as explained [in this comment](https://github.com/dreemurrs-embedded/Pine64-Arch/issues/263#issuecomment-1013928271). The performance gain was noticeable.

sorry-i-am-late, 2023-11-11:
> My PinePhone had lost both mobile and Wi-Fi connection a while ago and I became uninterested ever since.  

My phones wifi/bt chip has been dead for over a year and I think it is caused by voltage irregularities from boot looping while trying to charge  a completely dead battery.

Anyway, i did the VCCQ mod and it boots as it should after adding the files and recognizes the mode change but, i did not get any speed increase.  This is annoying because I overclocked my ram & cpu and if the emmc thing worked i was gonna do the gpu for an around the board speed boost.
I will probably try to revert it at some point, that way i can have jumpdrive and hopefully usb peripherals which also seem to have stopped working. (USB stuff was already sketchy since i have a ubports and i had to do the vconn mod to get them working at all)

dariox86, 2023-11-11:
I thought I was the only one affected by this hardware failure for connectivity. After a while, I remember having issues with USB, too. On the other hand, the performance gain brought by the modification was noticeable. Have you tried to benchmark it with hdparm?

sorry-i-am-late, 2023-11-11:
The hardware failure is not common but, there are a handful of others who have reported it. 

I tried hdparm and I get readings from the upper 50s to upper 80s (mostly 86ish).