dreemurrs-embedded/Pine64-Arch #264: [Plasma Mobile PinePhone]

hannescam, 2021-12-25:
cant chnge user from alarm to another account like "hannes"

dukla2000, 2022-01-04:
Not sure what steps you are trying nor what is failing for you.

But I use the following script - I have to disclose I haven't written 1 character of it myself, it is a copy/paste from (I think) the pinephone forum many months ago. So someone else deserves all the credit, not me!

`#!/bin/bash

# A script to change the current username

# Get current username
printf "\nType the current username:\n"
read old

# Set desired username
printf "\nType the desired username:\n"
read new

# Home paths
ohp="home/$old"
nhp="home/$new"

# Change username
printf "\nChanging username\n"

for file in group gshadow passwd shadow subgid subuid
do
    sed -i "s/$old/$new/g" /etc/$file*
done

# Rename home folder
mv /$ohp /$nhp

# Fix path references in /home for new user
printf "\nAdjusting paths in home directory\n"

grep -rl "$ohp" /$nhp | xargs sed -i "s+$ohp+$nhp+g"

# Set user info
echo
chfn $new
sync

reboot
`

Danct12, 2022-01-13:
Closing since this is not a bug nor enhancement. There's no plan to have a setup wizard (which supposedly have this feature) on first boot for now.

closed 2022-01-13 by Danct12