dreemurrs-embedded/Pine64-Arch #270: [waydroid not showing ui on sxmo]

Laar3, 2022-01-07:
- Device: PinePhone
- Kernel Version :  5.16.0-rc8-1-danctnix
- UI: sxmo (sway)

### Steps to reproduce
installing waydroid according to the instructions here https://github.com/dreemurrs-embedded/Pine64-Arch/wiki/Android-compatibility-layer then running `waydroid show-full-ui`

### Expected behavior
waydroid ui should open 

### Actual behavior
the terminal outputs 
```
[21:50:52] Starting waydroid session
[gbinder] Service manager /dev/binder has appeared
```
the terminal then just hangs

m-hugo, 2022-01-09:
Hi, I have the exact same problem with waydroid on sxmo on postmarketos on Oneplus 6 but it works fine with phosh.
same problem with `waydroid app launch someapp`.
also if I run `waydroid show-full-ui` while `waydroid session start` is running in another terminal, it spams `Failed to get service waydroidplatform, trying again...`

let me know if you get it to work

Laar3, 2022-01-09:
after some more testing i believe this is not a bug with sxmo or alarm but rather with waydroid, it seems to not work on systems with full disk encryption, it also happened on phosh aswell as across distros (pmos) and if disk encryption was enabled it would not work.

upon trying alarm without disk encryption though it worked straight away so i believe that this is a waydroid issue, not a alarm issue. Ill close the issue.

Laar3, 2022-01-09:
also for the record if anyone stumbles across this here are logs of what was happening so you can tell whether you have the same issue, i really should have included them in the initial issue but oh well https://pastebin.com/ED8KFGXB

closed 2022-01-09 by Laar3