dreemurrs-embedded/Pine64-Arch #273: Waydroid not working on sxmo

noeamiot, 2022-01-10:
- Device: Pinephone (1.2)
- Kernel Version : 5.16.0-rc8-1-danctnix
- UI: sxmo

### Steps to reproduce
Install the image from the [release](https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20211212/archlinux-pinephone-sxmo-20211212.img.xz)
The necessary  `pacman -Syu`
Follow instructions about [waydroid](https://github.com/dreemurrs-embedded/Pine64-Arch/wiki/Android-compatibility-layer)

### Expected behavior
`waydroid show-full-ui` shows something

### Actual behavior
`waydroid show-full-ui` hangs without complaining about anything

### Logfiles and additional information
`waydroid log` shows nothing interesting
Starting the container & session from a terminal doesn't give any useful information

`journalctl --folow` loops infinitly from a core dump related to vdso to errors about graphics.hwcomposer (Can't remember the exact name but if someone need it, I can simply reinstall the faulty versions and give the detail if someone need it).

### How to fix it
I read on the IRC in a question from Spagett awnsered by Osan saying that 2 packages had to be downgraded for it to work due to an upstream bug :

- sxmo-sway 1.6.0.1 -> 1.6.0 : [02374db2cda222ba35586829e05f6f7c3fdbf9d9](https://github.com/dreemurrs-embedded/Pine64-Arch/commit/02374db2cda222ba35586829e05f6f7c3fdbf9d9)
- sxmo-wlroot 0.14.0.0 -> 0.13.0.0 : [d0a0fd5d5082b3fd5b535576ab8cd6234e1924f2](https://github.com/dreemurrs-embedded/Pine64-Arch/commit/d0a0fd5d5082b3fd5b535576ab8cd6234e1924f2)

So I did it and voila, waydroid is now working perfectly in sxmo with arch !

May or may not be related to #270 as the logs given were not the same

I first wrote this issue so that people with have the same problem can find the solution as well (as the IRC does not show up in browser searches ;o) but maybe I should make a PR to revert those two commits while waiting for the upstream fix, I wait for your feedback on this last point !

Danct12, 2022-01-11:
This is somewhat a weird bug, so far I'm not sure why this happened.

noeamiot, 2022-01-19:
I did a few tests and here is the thing : `pacman -S sway` (to replace sxmo-sway by sway and sxmo-wlroots by wlroots) also fixed this issue.

The detail we may have missed about the packages downgrading is :
- Both sxmo-wlroots 0.13.0 and 0.14.0 are in fact wlroots v 0.15.0 (by reading the meson.build in sxmo-wlroots code on sr.ht)
- Both sxmo-sway 1.6.0 and 1.6.0.1 are in fact sway 1.6.0 (same as above)

On the [official releases of sway 1.6.0](https://github.com/swaywm/sway/releases/tag/1.6) they say that for sway 1.6.0 it is recommended to use wlroots 0.13.0 which may be why we have this issue.
On the other hand the current community package for sway is version 1.6.1 and 0.14.1 for wlroots. On the [official releases of sway 1.6.1](https://github.com/swaywm/sway/releases/tag/1.6.1) they say that this version is compatible with wlroots 0.14.

The only thing I can't explain with this is why the sxmo-sway (corresponding to sway 1.6.0) does not have the same bug when used with sxmo-wlroots 0.13.0 (which is on the same wlroots than 0.14.0, the 0.15.0 one).

ArenM, 2022-01-20:
This seems like a recent regression in sway or wlroots, if you build sway-git from the aur it's also broken.

The wlroots and sway versions are tagged from the upstream git HEAD by the sxmo people, I'm not really sure why they picked the version numbers they did. You're correct that sxmo-wlroots is more like wlroots 0.15, but the sway version is also more like sway 1.7. Neither of them are an official release.

Installing the sway package is definitely a simpler workaround, now that it's available. That is one of reasons I changed sxmo-utils to depend on sway in #271. You should be aware the purpose of sxmo-sway is to fix some touch handling issues, so the menus might not work correctly.

anjandev, 2022-02-17:
FWIW, I can reproduce this bug on the latest Sxmo in postmarketOS stable (v21.12 service pack 2).

anjandev, 2022-03-07:
Also related: https://github.com/waydroid/waydroid/issues/294

hartgeld, 2023-10-10:
Any updates for waydroid running on sxmo arch mobile? Device is PinephonePro 
dmesg as well as waydroid logcat is complaining: "Could not find 'android.hardware.graphics.composer@2.1::IComposer/default' for ctl.interface_start from pid: 21 (/system/bin/hwservicemanager)"