dreemurrs-embedded/Pine64-Arch #277: Unable to enter full disk encryption with physical keyboard

daniellejackson-dialexa, 2022-01-12:
- Device: Pinephone and Pinetab
- Kernel Version : 5.16.0-1-danctnix
- UI: phish

### Steps to reproduce
- Boot with full disk encryption and keyboard attached

### Expected behavior
- Key can be entered and device unlocked

### Actual behavior
- On screen keyboard will be hidden
- Keypresses do not register in text field, pressing return does not start unlocking
- "OSK" button does not display on screen keyboard, no reaction when tapping it
- Happens with either the pinetab docked keyboard, pinephone keyboard case, and USB attached keyboard on either. 

### Logfiles and additional information
- Can't log in that far to get the log

Danct12, 2022-01-13:
Known bug: https://gitlab.com/postmarketOS/pmaports/-/issues/1383

luigi311, 2022-01-22:
Looks like its been solved upstream with only minor changes. Heres the PR
https://gitlab.com/postmarketOS/pmaports/-/merge_requests/2854

Danct12, 2022-01-24:
Fixed in 8174f444bf60c48e8d02a49590b88d40d3424d00

closed 2022-01-24 by Danct12