dreemurrs-embedded/Pine64-Arch #284: gnome-calendar-mobile "New Event..." does not open

mikeshenson, 2022-01-22:
- Device: PinePhone 32G
- Kernel Version : 5.16.1-2-danctnix
- UI: Phosh 
- gnome-calendar-mobile 41.2-1

### Steps to reproduce
Month view >> Click on the "+2" >> Press New Event... >> Nothing happens

### Expected behavior
I expect the new event window to open

### Actual behavior
Nothing happens

[image]

mikeshenson, 2022-02-12:
I created an upstream issue here
https://gitlab.gnome.org/GNOME/gnome-calendar/-/issues/797

closed 2022-02-12 by mikeshenson

Danct12, 2022-02-13:
Yes, GNOME Calendar was not adaptive on mobile by default. This is where that Purism repo comes into place, we get the patches from there and use them.

mikeshenson, 2022-02-13:
On my phone, I uninstalled gnome-calendar-mobile and installed the gnome-calendar from extra. 
I could reproduce this issue with gnome-calendar from extra, this is why I closed the issue here and opened it up with gnome.
[image]
.

mikeshenson, 2022-02-13:
So... 
Issue #285 and Issue #286 should be moved to source.puri.sm?