dreemurrs-embedded/Pine64-Arch #292: Improve documentation by explaining how to add support for eMMC modification

dariox86, 2022-01-28:
- Device: PinePhone
- Kernel Version : 5.16.2-2-danctnix
- UI: Phosh

### Steps to reproduce
Installing Arch Linux ARM DanctNIX on a PinePhone with a modified eMMC does not work out of the box

### Expected behavior
A note in the wiki explaining what to do to have a modified PinePhone to boot. I can send a pull request.

### Actual behavior
A modified PinePhone does not work out of the box.

Danct12, 2022-01-30:
I was thinking of packing up vccq.zip and tell people how to apply it, since it's not possible to apply it within the system thus no point to make a package for that.

BTW, the wiki is open, you can make a new page for that.

dariox86, 2022-01-30:
I can take care of the wiki. What link should I supply to point out the relevant files to add?

xnopasaranx, 2022-04-06:
I would love to have that information right now... any chance you could add that wiki page @dariox86 ?

dariox86, 2022-04-06:
At the moment I am at work. Basically [this](https://github.com/dreemurrs-embedded/Pine64-Arch/issues/263#issuecomment-1013928271) is everything you need to do.

Danct12, 2024-08-29:
https://github.com/dreemurrs-embedded/Pine64-Arch/discussions/656

closed 2024-08-29 by Danct12