dreemurrs-embedded/Pine64-Arch #293: Blackscreen after update on Plasma

Dan-Sun, 2022-01-29:
- Device: PinePhone
- Kernel Version : 5.16.3-2-danctnix
- UI: Plasma

### Steps to reproduce
Update the latest image (24.1.22)

### Expected behavior
No blackscreen

### Actual behavior
Blackscreen

### Logfiles and additional information
```
Jan 29 21:55:31 danctnix systemd-logind[2982]: Power key pressed short.
Jan 29 21:55:33 danctnix kernel: kb151 2-0015: Charger read failed - MCU returned 0xff
Jan 29 21:55:33 danctnix kernel: kb151 2-0015: Failed to initialize the charger
Jan 29 21:55:34 danctnix systemd-logind[2982]: Power key pressed short.
Jan 29 21:55:36 danctnix dbus-daemon[3213]: [session uid=1000 pid=3213] Activating service name='org.freedesktop.Notifications' requested by ':1.38' (uid=1000 pid=4208 comm="/usr/lib/kscreenlocker_greet --immediateLock --gra")
Jan 29 21:55:36 danctnix dbus-daemon[4217]: writing oom_score_adj error: Permission denied
Jan 29 21:55:36 danctnix kscreenlocker_greet[4208]: kf.kirigami: Units.devicePixelRatio is deprecated (since 5.86): This returns 1 when using Qt HiDPI scaling.
Jan 29 21:55:36 danctnix kscreenlocker_greet[4208]: kf.kirigami: Units.wheelScrollLines is deprecated (since 5.86): Use Qt.styleHints.wheelScrollLines instead
Jan 29 21:55:36 danctnix kscreenlocker_greet[4208]: WARNING: Cannot find style "org.kde.breeze" - fallback: "/usr/lib/qt/qml/QtQuick/Controls/Styles/Base"
Jan 29 21:55:37 danctnix systemd-logind[2982]: Power key pressed short.
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: Both point size and pixel size set. Using pixel size.
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/indicators/SignalStrength.qml:19: ReferenceError: strengthLabel is not defined
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/indicators/SignalStrength.qml:20: ReferenceError: strengthLabel is not defined
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/LockScreen.qml:53: ReferenceError: wallpaper is not defined
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: Both point size and pixel size set. Using pixel size.
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/indicators/SignalStrength.qml:19: ReferenceError: strengthLabel is not defined
Jan 29 21:55:37 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/indicators/SignalStrength.qml:20: ReferenceError: strengthLabel is not defined
Jan 29 21:55:38 danctnix kscreenlocker_greet[4208]: Both point size and pixel size set. Using pixel size.
Jan 29 21:55:38 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/indicators/SignalStrength.qml:19: ReferenceError: strengthLabel is not defined
Jan 29 21:55:38 danctnix kscreenlocker_greet[4208]: file:///usr/share/plasma/look-and-feel/org.kde.plasma.phone/contents/lockscreen/indicators/SignalStrength.qml:20: ReferenceError: strengthLabel is not defined
Jan 29 21:55:38 danctnix kscreenlocker_greet[4208]: qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
Jan 29 21:55:38 danctnix kscreenlocker_greet[4208]: qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
Jan 29 21:55:38 danctnix kernel: kb151 2-0015: Charger read failed - MCU returned 0xff
Jan 29 21:55:38 danctnix kernel: kb151 2-0015: Failed to initialize the charger
Jan 29 21:55:44 danctnix kernel: kb151 2-0015: Charger read failed - MCU returned 0xff
Jan 29 21:55:44 danctnix kernel: kb151 2-0015: Failed to initialize the charger
Jan 29 21:55:44 danctnix dbus-daemon[2976]: [system] Activating via systemd: service name='org.freedesktop.home1' unit='dbus-org.freedesktop.home1.service' requested by ':1.74' (uid=1000 pid=4228 comm="kcheckpass -m classic -S 44 ")
Jan 29 21:55:44 danctnix dbus-daemon[2976]: [system] Activation via systemd failed for unit 'dbus-org.freedesktop.home1.service': Unit dbus-org.freedesktop.home1.service not found.
Jan 29 21:55:44 danctnix kcheckpass[4228]: pam_systemd_home(kde:auth): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Jan 29 21:55:44 danctnix audit[4230]: USER_AUTH pid=4230 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:unix_chkpwd acct="alarm" exe="/usr/bin/unix_chkpwd" hostname=? addr=? terminal=? res=success'
Jan 29 21:55:44 danctnix kernel: audit: type=1100 audit(1643493344.299:156): pid=4230 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:unix_chkpwd acct="alarm" exe="/usr/bin/unix_chkpwd" hostname=? addr=? terminal=? res=success'
Jan 29 21:55:44 danctnix kscreenlocker_greet[4208]: org.kde.plasma.pulseaudio: No object for name "alsa_output.platform-sound.HiFi__hw_PinePhone_0__sink"
Jan 29 21:55:44 danctnix kscreenlocker_greet[4208]: org.kde.plasma.pulseaudio: No object for name "alsa_output.platform-sound.HiFi__hw_PinePhone_0__sink"
Jan 29 21:55:44 danctnix kscreenlocker_greet[4208]: org.kde.plasma.pulseaudio: No object for name "alsa_input.platform-sound.HiFi__hw_PinePhone_0__source"
Jan 29 21:55:49 danctnix kernel: kb151 2-0015: Charger read failed - MCU returned 0xff
Jan 29 21:55:49 danctnix kernel: kb151 2-0015: Failed to initialize the charger
Jan 29 21:55:54 danctnix kernel: kb151 2-0015: Charger read failed - MCU returned 0xff
Jan 29 21:55:54 danctnix kernel: kb151 2-0015: Failed to initialize the charger
Jan 29 21:55:59 danctnix kernel: kb151 2-0015: Charger read failed - MCU returned 0xff
Jan 29 21:55:59 danctnix kernel: kb151 2-0015: Failed to initialize the charger

```

Danct12, 2022-01-30:
Known upstream bug: https://invent.kde.org/teams/plasma-mobile/issues/-/issues/108

Danct12, 2022-01-30:
I've forked kactivitymanagerd and applied one of the patch for fixing the issue: b9b43ac0bcb7d8a40a9552b1a9c367487263bc44

closed 2022-01-30 by Danct12