dreemurrs-embedded/Pine64-Arch #294: Please document how to disable splash (animated) screen on boot

oz123, 2022-01-29:
- Device: PinePhone
- Kernel Version : 5.12.14-1-danctnix
- UI:  phose

### Steps to reproduce
I did the following steps: remove splash lines from `boot/boot.txt`
```
$ grep env /boot/boot.txt 
	setenv linux_mmcdev 0
	setenv linux_mmcdev 2
	setenv rootpart 2
	setenv rootpart 1
	setenv bootdir "/boot"
setenv bootargs loglevel=4 console=${console} console=tty0 root=/dev/mmcblk${linux_mmcdev}p${rootpart} rw rootwait
setenv ramdisk_size ${filesize}
setenv user_scriptaddr 0x61dbc200
```

I also removed the package `bootsplash-theme-danctnix

I then ran:
```
sudo mkinitcpio -P
==> Building image from preset: /etc/mkinitcpio.d/linux-megi.preset: 'default'
  -> -k 5.16.3-2-danctnix -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: 5.16.3-2-danctnix
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [keyboard]
  -> Running build hook: [osk-sdl]
  -> Running build hook: [fsck]
==> ERROR: Hook 'bootsplash-danctnix' cannot be found
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-linux.img
==> WARNING: errors were encountered during the build. The image may not be complete
```
Followed by
```
$ cd /boot/
[alarm@danctnix boot]$ ./mkscr 
mkimage: Can't open boot.scr: Permission denied
[alarm@danctnix boot]$ sudo ./mkscr 
Image Name:   U-Boot boot script
Created:      Sat Jan 29 23:29:31 2022
Image Type:   ARM Linux Script (uncompressed)
Data Size:    1363 Bytes = 1.33 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 1355 Bytes = 1.32 KiB = 0.00 MiB
```
### Expected behavior

The phone uses text messages at boot.

### Actual behavior
I still see the animated splash at boot.

Any help would be appreciated.

Dan-Sun, 2022-01-30:
Hey 

I also wanted to get rid of it earlier and did it that way:
```
pacman -Rsn bootsplash-theme-danctnix
```
```
nano /etc/mkinitcpio.conf
```
```
HOOKS=(base udev autodetect modconf block filesystems keyboard osk-sdl fsck)
```
```
nano /boot/boot.txt
```
Delete (quiet is up 2 you):
```
quiet bootsplash.bootfile=bootsplash-themes/danctnix/bootsplash
```
```
mkimage -A arm -O linux -T script -C none -n "U-Boot boot script" -d boot.txt boot.scr
```
```
mkinitcpio -P
```

oz123, 2022-01-30:
Hi Dan! 
Thank you for your reply.
I tried all the steps you showed. After reboot, I am still seeing the splash image.
Here is a log of actions:

```
[alarm@danctnix ~]$ sudo pacman -Rsn bootsplash-theme-danctnix
[sudo] password for alarm: 
error: target not found: bootsplash-theme-danctnix
[alarm@danctnix ~]$ cd /boot/
[alarm@danctnix boot]$ grep HOOKS /etc/mkinitcpio.conf
# HOOKS
# This is the most important setting in this file.  The HOOKS control the
# order in which HOOKS are added.  Run 'mkinitcpio -H <hook name>' for
#    HOOKS=(base)
#    HOOKS=(base udev autodetect block filesystems)
#    HOOKS=(base udev block filesystems)
#    HOOKS=(base udev block mdadm encrypt filesystems)
#    HOOKS=(base udev block lvm2 filesystems)
# HOOKS=(base udev autodetect modconf block filesystems keyboard osk-sdl fsck bootsplash-danctnix)
HOOKS=(base udev autodetect modconf block filesystems keyboard osk-sdl fsck)
[alarm@danctnix boot]$ grep setenv /boot/boot.txt 
	setenv linux_mmcdev 0
	setenv linux_mmcdev 2
	setenv rootpart 2
	setenv rootpart 1
	setenv bootdir "/boot"
setenv bootargs loglevel=4 console=${console} console=tty0 root=/dev/mmcblk${linux_mmcdev}p${rootpart} rw rootwait
setenv ramdisk_size ${filesize}
setenv user_scriptaddr 0x61dbc200
[alarm@danctnix boot]$ sudo su
[root@danctnix boot]# mkimage -A arm -O linux -T script -C none -n "U-Boot boot script" -d boot.txt boot.scr
Image Name:   U-Boot boot script
Created:      Sun Jan 30 14:34:54 2022
Image Type:   ARM Linux Script (uncompressed)
Data Size:    1363 Bytes = 1.33 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 1355 Bytes = 1.32 KiB = 0.00 MiB
[root@danctnix boot]# rm boot.scr 
[root@danctnix boot]# mkimage -A arm -O linux -T script -C none -n "U-Boot boot script" -d boot.txt boot.scr
Image Name:   U-Boot boot script
Created:      Sun Jan 30 14:35:00 2022
Image Type:   ARM Linux Script (uncompressed)
Data Size:    1363 Bytes = 1.33 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 1355 Bytes = 1.32 KiB = 0.00 MiB
[root@danctnix boot]# less boot.scr 
"boot.scr" may be a binary file.  See it anyway? 
[root@danctnix boot]# mkinitcpio -P
==> Building image from preset: /etc/mkinitcpio.d/linux-megi.preset: 'default'
  -> -k 5.16.3-2-danctnix -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: 5.16.3-2-danctnix
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [keyboard]
  -> Running build hook: [osk-sdl]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-linux.img
==> Image generation successful
[root@danctnix boot]# reboot
Connection to 192.168.1.122 closed by remote host.
Connection to 192.168.1.122 closed.
```

I have all these files in `/boot/` and I was thinking, maybe I still see the image because I am using the FDE image?

```
[alarm@danctnix ~]$ ls -l /boot/
total 108038
-rw-r--r-- 1 root root     1427 Jan 30 14:35 boot.scr
-rw-r--r-- 1 root root     1355 Jan 29 23:26 boot.txt
drwxr-xr-x 4 root root     3488 Jan 27 23:43 dtbs
-rw-r--r-- 1 root root 19939336 Jan 28 21:45 Image
-rw-r--r-- 1 root root  8206361 Jan 28 21:45 Image.gz
-rw------- 1 root root 77551143 Jan 30 14:36 initramfs-linux.img
-rwxr-xr-x 1 root root      239 Jan 16 20:19 mkscr
-rw-r--r-- 1 root root   745711 Jan 16 20:19 u-boot-sunxi-with-spl-pinephone-492.bin
-rw-r--r-- 1 root root   745711 Jan 16 20:19 u-boot-sunxi-with-spl-pinephone-528.bin
-rw-r--r-- 1 root root   745711 Jan 16 20:19 u-boot-sunxi-with-spl-pinephone-552.bin
-rw-r--r-- 1 root root   745711 Jan 16 20:19 u-boot-sunxi-with-spl-pinephone-624.bin
-rw-r--r-- 1 root root   701843 Jan 16 20:19 u-boot-sunxi-with-spl-pinetab-492.bin
-rw-r--r-- 1 root root   701843 Jan 16 20:19 u-boot-sunxi-with-spl-pinetab-528.bin
-rw-r--r-- 1 root root   701843 Jan 16 20:19 u-boot-sunxi-with-spl-pinetab-552.bin
-rw-r--r-- 1 root root   701843 Jan 16 20:19 u-boot-sunxi-with-spl-pinetab-624.bin
```
Do you have all these files too?

oz123, 2022-01-30:
Following this thread, https://forum.pine64.org/showthread.php?tid=15225, could it be that all the output is forced to go to the serial console?

bigbiggie, 2022-01-30:
I'm using an FDE image and the instructions provided by Dan worked for me. Sorry, if this doesn't help, just wanted to point it out.

oz123, 2022-01-30:
It seems that plugging a keyboard and switching a TTY with CTL+F2 is the way to see the messages after all the steps above.

Dan-Sun, 2022-02-01:
Hm... Just did a fresh fde install - it works.

Danct12, 2023-08-29:
The bootsplash has since been removed in linux-megi 6.4.10, closing this.