dreemurrs-embedded/Pine64-Arch #296: Top row of PP keyboard has a wrong layout

bigbiggie, 2022-01-30:
- Device: (PinePhone or PineTab?) Pinephone CE 3GB
- Kernel Version : $(uname -r) 5.16.3-2-danctnix
- UI: Phosh

### Steps to reproduce

Attach PinePhone keyboard to the PinePhone, boot Arch Linux Arm, and try the top row keys using FN-keys. The keys are not mapped correctly. Same keys use the same key code (such as FN+1 and FN+2) and I cannot use the function keys (for example to switch to another virtual terminal).

### Expected behavior

FN+1 = |
FN+2 = \
FN+3 = $
etc.

### Actual behavior

FN+1 = \
FN+2 = \
FN+3 = £
etc.

### Logfiles and additional information

Will provide it, if necessary

I read that the driver available in the AUR (https://aur.archlinux.org/packages/pinephone-keyboard-git/) works much better, but since the kernel driver is a builtin, I cannot use the driver from the AUR.

Danct12, 2022-02-20:
BTW, the keyboard input driver can be disabled by adding `kb151.disable_input` to the kernel boot arguments.

bigbiggie, 2022-02-21:
That worked, thank you. I'm using the keyboard right now.