dreemurrs-embedded/Pine64-Arch #301: HDMI docking output displays password

Dan-Sun, 2022-02-01:
- Device: PinePhone
- Kernel Version : 5.16.4-1-danctnix
- UI: Plasma

### Steps to reproduce
Install latest encrypted image. Connect device to dock and type your password.

### Expected behavior
No password visible. 

### Actual behavior
Password visible.

Danct12, 2022-02-07:
Does it also show the password in raw text? If so, that seems to be a upstream issue and worth reporting: https://invent.kde.org/plasma/plasma-mobile

Dan-Sun, 2022-02-07:
Sorry it has nothing to do with the UI in general. It's the encryption (fde).

hamblingreen1, 2022-09-06:
reproduced with the latest fde build. likely an upstream issue with osk-sdl, that it doesnt render on a second monitor.