dreemurrs-embedded/Pine64-Arch #303: Cyclic makedepends prevents building toolchain

3nprob, 2022-02-04:
### Steps to reproduce
Attempt to build the packages in this repository via `build.sh` with this makeshift script (yes it could be a lot cleaner, should prob be a Makefile):

```
#!/bin/bash
for _repo in PKGBUILDS/*; do
        repo=$(echo $_repo | sed "s#PKGBUILDS/##")
        mkdir -p "./repo/${repo}/aarch64"
        sudo mkdir -p "chroot-aarch64/user/home/user/dev/Pine64-Arch/repo/${repo}/aarch64/"
        sudo mkdir -p "chroot-aarch64/root/home/user/dev/Pine64-Arch/repo/${repo}/aarch64/"
        repo-add ./repo/${repo}/aarch64/${repo}.db.tar.gz ./repo/${repo}/aarch64/*.pkg.tar.xz

        sudo rsync -pHAXa repo/ chroot-aarch64/user/repo/
        sudo rsync -pHAXa repo/ chroot-aarch64/root/repo/
        sudo rsync -pHAXa repo/ "chroot-aarch64/root/home/user/dev/Pine64-Arch/repo/${repo}/"
        sudo rsync -pHAXa repo/ "chroot-aarch64/user/home/user/dev/Pine64-Arch/repo/${repo}/"
        for _pkg in ${_repo}/*; do
                pkg=$(echo $_pkg | sed "s#PKGBUILDS/${repo}/##")
                ./build.sh --arch aarch64 --repo ${repo} --pkg ${pkg} || true
                sudo rsync -pHAXa repo/ chroot-aarch64/user/repo/
                sudo rsync -pHAXa repo/ chroot-aarch64/root/repo/
                sudo rsync -pHAXa repo/ "chroot-aarch64/root/home/user/dev/Pine64-Arch/repo/${repo}/"
                sudo rsync -pHAXa repo/ "chroot-aarch64/user/home/user/dev/Pine64-Arch/repo/${repo}/"
        done
done
```

### Expected behavior
Can build all packages as long as they're built in the right order.

### Actual behavior

There is a cyclic `makedepends` between [`arm-none-eabi-newlib`](https://github.com/dreemurrs-embedded/Pine64-Arch/blob/a207c6caac087950075ba116509f6554056d4907/PKGBUILDS/danctnix/arm-none-eabi-newlib/PKGBUILD#L13)` and [`arm-none-eabi-gcc`](https://github.com/dreemurrs-embedded/Pine64-Arch/blob/a207c6caac087950075ba116509f6554056d4907/PKGBUILDS/danctnix/arm-none-eabi-gcc/PKGBUILD#L16).

So it will fail at either/both of:

```
$ ./build.sh --arch aarch64 --repo danctnix --pkg arm-none-eabi-newlib
$ ./build.sh --arch aarch64 --repo danctnix --pkg arm-none-eabi-gcc
```

Danct12, 2022-02-07:
I'm actually not too sure how this was even compiled in the first place. :confused: 

If you check upstream, the same thing is happening over there too:
- https://archlinux.org/packages/community/any/arm-none-eabi-newlib/
- https://archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/

As you can see, both packages are makedepends of each other. So in the end what I did was taking arm-none-eabi-newlib from upstream repository as it is architecture-independent and then compile arm-none-eabi-gcc.

Danct12, 2024-09-28:
We no longer provide the toolchain above (and haven't been for years). Closing.

closed 2024-09-28 by Danct12