dreemurrs-embedded/Pine64-Arch #305: left-earphone only sound on headphones during calls

lafleurdeboum, 2022-02-05:
- Device: PinePhone
- Kernel Version : 5.16.4-1-danctnix
- UI: Phosh

### Steps to reproduce
  - plug headphones in
  - call eg the answering machine
  - state that the sound appears only in the left earphone

### Expected behavior
Have the sound dispatched on both earphones.

### Actual behavior
The sound appears only in the left headphone. I also noticed that the sound rockers produce a sound that _is_ dispatched on both earphones during calls - so the `call` profile itself is certainly not causing this.

### Logfiles and additional information
See the [initial issue at Phosh](https://gitlab.gnome.org/World/Phosh/phosh/-/issues/685). Users on other phones/distros seem to not have this issue.

`gnome-calls -l 123 -vvvvv` output : 
```
02:30:08.7858             GLib-GIO[ 4455]:   DEBUG: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
02:30:08.7863                dconf[ 4455]:   DEBUG: watch_fast: "/org/gnome/Calls/" (establishing: 0, active: 0)
02:30:08.7887        CallsSettings[ 4455]:   DEBUG: Enabling the use of default origins
02:30:08.7891                dconf[ 4455]:   DEBUG: change_fast
02:30:08.7906        CallsSettings[ 4455]:   DEBUG: Setting country code to FR
02:30:08.7912                dconf[ 4455]:   DEBUG: change_fast
02:30:08.7924                dconf[ 4455]:   DEBUG: change_fast
02:30:08.7933                dconf[ 4455]:   DEBUG: change_fast
02:30:08.7939                dconf[ 4455]:   DEBUG: change_fast
02:30:08.7943                dconf[ 4455]:   DEBUG: change_fast
02:30:08.7939                dconf[ 4455]:   DEBUG: watch_established: "/org/gnome/Calls/" (establishing: 1)
02:30:08.7995                dconf[ 4455]:   DEBUG: watch_fast: "/org/freedesktop/folks/" (establishing: 0, active: 0)
02:30:08.8056                dconf[ 4455]:   DEBUG: watch_established: "/org/freedesktop/folks/" (establishing: 1)
02:30:08.8332             GLib-GIO[ 4455]:   DEBUG: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
02:30:08.8378         CallsManager[ 4455]:   DEBUG: Scanning for plugins in `/usr/lib/calls/plugins'
02:30:08.8382     CallsDBusManager[ 4455]:   DEBUG: Registering at /org/gnome/Calls
02:30:08.8715                dconf[ 4455]:   DEBUG: sync
```

Danct12, 2022-02-11:
Can you tell if the issue still persist in the latest update?

lafleurdeboum, 2022-02-12:
Thus issue is now fixed ! Thank you very much !
👍️

closed 2022-02-13 by Danct12