dreemurrs-embedded/Pine64-Arch #307: Update warning, "Warning /etc/sudoers installed as /etc/sudoers.pacnew"

InfiniteOtter, 2022-02-06:
- Device: (PinePhone or PineTab?)
- Kernel Version : archlinux-pinephone-pro-phosh-20220124

### Steps to reproduce
I installed Arch phosh and changed the password for Alarm.
When I ran "sudo pacman -Syu" I got the error, "Warning /etc/sudoers installed as /etc/sudoers.pacnew."

### Expected behavior
I think this is the expected behavior under the circumstances.

### Actual behavior
Since the update was kind and didn't overwrite my sudo password list, I am not sure if I need to update any passwords that may be in the new file and I am not sure how to merge things between these files if I do.

### Desired Behavior
If this is or is not an issue, I think a note on the Installation Guide https://github.com/dreemurrs-embedded/Pine64-Arch/wiki/Installation-Guide  might give those of us users who pay attention to the log outputs a bit more guidance and comfort.

DrRac27, 2022-02-06:
This is the expected behavior and already documented in the wiki:
https://wiki.archlinux.org/title/Pacman/Pacnew_and_Pacsave

Danct12, 2022-02-06:
Not a bug. See URL above ^

closed 2022-02-06 by Danct12